roman-corgi/corgidrp #243: Boresight calibration for defocused PSFs

mygouf, 2024-10-30:
Boresight calibration should work with at least 200nm defocused PSFs (V2 to be delivered as part of R3.0.2)

nasavbailey, 2024-11-05:
First light defocus could be ~300-500nm. I believe John provided PSFs at this range of defocus, and we should make sure we have a solution for this case.

semaphoreP, 2024-11-05:
Just to clarify, by "should work", does that mean we should be measuring the boresight offset to within requirements, even with 500 nm of defocus?

nasavbailey, 2024-11-06:
Good point. We haven't written a requirement for first light. I think our performance could be relaxed, since at first light, the observatory WFE doesn't meet its requirement either. We just need to be good enough to subsequently place an isolated star near the center of the FOV for the subsequent CGI focus and alignment activity. To throw out a number: how's 100mas (~5px) 1 sigma? We could probably relax further if needed.

nasavbailey, 2024-12-03:
After some further discussions at JPL, we have elected to change our first light plan to include a focus sweep so that DRP doesn't need to develop new functionality. (It'll also have the added benefit of making the subsequent phase retrieval easier) At first light, we will still observe the boresight field, but now we will take multiple sets of images, each with CGI's focus control mirror (FCM) at a different focus value, with focus steps of ~100nm. 

The DRP would only need to produce "close enough" results on the best-focused image of this first light set. That said, given the nominal step size, we'd be a maximum of 50nm from best focus, with maybe another (ballpark) 75nm of other aberrations. @manduhmia, I remember you did some performance testing on defocused simulated images. Does DRP meet the requirement for, say, the 100nm defocus simulated image, as a proxy for rms[50nm defocus, 75nm other aberrations]?

I think this means we can ditch the R&D on improved algorithms to handle strongly defocused images (ie: don't have to do the work originally requested in this ticket!). But this dataset will be off-nominal for astrometry (N sets of M images). Does this necessitate manual image sorting and/or a helper script for the first light activity only? We can't know ahead of time which image set will be closest to focus. 

A corollary - in order for someone (could be us could be CTC. Thoughts?) to use these datasets to estimate the best focus value for the FCM, they'll need clean L2b data products.  Does the astrometry module save the L2a and L2b products? If not, can that be a new issue for R3.0.2?

semaphoreP, 2025-03-17:
@manduhmia can we close the loop here and assert if:
 1. the current code can handle defocus of 100 nm rms
 2. the astrometry recipe saves the L2a and L2b products

manduhmia, 2025-03-21:
Current code can handle 100nm defocus to the extent that: plate scale is within 0.5 [mas], north angle within 0.05 [deg], and boresight correct within (-8.72246005e-06, -3.58430322e-08) [deg] although the requirement is to be within (8.333e-7, 8.333e-7) [deg].

manduhmia, 2025-03-21:
Also, the astrometry recipe saves L2a products on line 36 and save L2b products on line 90.

semaphoreP, 2025-03-27:
From Amanda's findings above: we are recovering boresight to within 32 mas (better than 100 mas). The main source of bias is that John's sims and Amanda's sims have the star's y-positions (I think y) in different pixels if you compare what is the their theoretical positions. The x-axis seems to be fine though, and the boresight is found to < mas levels of precision. Regardless, seems like we are good there.

And Amanda confirms the boresight recipe saves L2a and L2b files. So I think this can be closed. Please reopen is you disagree.

closed 2025-03-27 by semaphoreP