roman-corgi/corgidrp #244: nonlinearity calibration product has unnecessary DQ header

kjl0025, 2024-10-30:
**Describe the bug**
A DQ header is created for the nonlinearity calibration product.  However, the array values in the nonlinearity array are relative gain values for specific EM gain-DN combinations (not pixel data).  So I don't think we need a DQ array for this.  Its presence doesn't hurt anything, but a user new to corgidrp may be confused to find this header in the calibration product that doesn't have any meaning.

**To Reproduce**
Steps to reproduce the behavior:
For example, run an e2e test or unit test that creates a nonlinearity calibration product.

**Expected behavior**
An array of zeros should be produced for the 3rd extension header (the default DQ header).

**corgidrp version:**
Main branch, v1.0

**Additional context**
Should be a simple fix to remove the DQ header from the NonlinCalibration class.

kjl0025, 2024-10-30:
Same applies for the astrometric calibration product.

semaphoreP, 2024-10-31:
Are you saying the current DQ frame isn't an array of all 0's, or are you requesting the DQ extension to be removed completely.

Removing it may make it harder to subclass these data classes from the Image subclass, so I'm inclined to keep the DQ extension for simplicity.

kjl0025, 2024-11-06:
I was suggesting removing the DQ extension completely to avoid confusion.  But I agree with you that it would be simpler to keep it, which would ensure good compatibility with `Image`.  I don't have strong feelings on this one, just making folks aware of it.  Maybe we could put a comment in the headers for these calibration classes that the DQ is not meaningful?

semaphoreP, 2024-11-06:
Sure, sounds good to me.

kjl0025, 2024-11-06:
Cool, I'll add a comment to the relevant headers for my next PR.

kjl0025, 2025-05-15:
The comment about the irrelevance of the DQ header in these cases has been implemented, so we can close this issue.

closed 2025-05-15 by kjl0025