roman-corgi/corgidrp #245: Should we remove the header indicating data level from calibration products?

kjl0025, 2024-10-30:
**Is your feature request related to a problem? Please describe.**
The data level for a calibration product, designated in the header, may be confusing.  It can represent to some degree what steps have been taken to process the raw data used for calibration, but it may also be confusing since not all steps per level are actually applied depending on the product.

**Describe the solution you'd like**
Remove the data level designation in each calibration data class in data.py.  And the steps taken to process the data should be present in the "History" in the header info.

**Additional context**
--

semaphoreP, 2024-10-30:
@agreenbaum, any opinion on how we should modify the data level for calibration products? We do have other header keywords that indicate it's a calibration product, so not sure how important it is the calibration products have or don't have a data level.

maxwellmb, 2024-10-30:
Perhaps worth noting that the current filenaming convention includes a
"level" in the filename too. Could we just replace all the levels (both in
the headers and in the filename) with "CAL" or something like that?

agreenbaum, 2024-10-31:
There are some notes on calibration products under construction: https://collaboration.ipac.caltech.edu/pages/viewpage.action?pageId=113907541 (See Calibration Data Products sub-heading)
Maybe data level could be "CAL" just like the file name suffix? @mygouf -- have you and others thought about this already?

mygouf, 2024-10-31:
Having "CAL" as a data level in the header sounds like a reasonable suggestion Alex. I like that.

semaphoreP, 2024-10-31:
I like it as well. Let's aim to have this for v2.0

closed 2025-03-27 by semaphoreP