roman-corgi/corgidrp #250: How to run e2e tests with the changes in obstype-switch? "No valid {0} calibration in caldb"

hsergi, 2024-11-07:
**Describe the bug**
I have completely deleted ``.corgidrp/``
I have completely deleted corgidrp repo in my computer
I download a fresh copy of the repo
I run ``pip install -e .`` successfully
I switch to branch ``obstype-switch``
I run the e2e for nonlin: python nonlin_e2e.py
The run crashes:

`corgidrp: Configuration file written to /Users/srhildeb/.corgidrp/corgidrp.cfg. Please edit if you want things stored in different locations.
Group of 30 files starting with CGI_EXCAM_L1_0000051841.fits
Group of 110 files starting with CGI_EXCAM_L1_0000051731.fits
Group of 44 files starting with CGI_EXCAM_L1_0000051941.fits
Group of 66 files starting with CGI_EXCAM_L1_0000051986.fits
Group of 66 files starting with CGI_EXCAM_L1_0000055122.fits
Group of 66 files starting with CGI_EXCAM_L1_0000055191.fits
WARNING: VerifyWarning: Card is too long, comment will be truncated. [astropy.io.fits.card]
Running walker
Traceback (most recent call last):
  File "nonlin_e2e.py", line 240, in <module>
    test_nonlin_cal_e2e(args.tvacdata_dir, args.output_dir)
  File "nonlin_e2e.py", line 167, in test_nonlin_cal_e2e
    walker.walk_corgidrp(nonlin_l1_list, '', e2eoutput_path)
  File "/Users/srhildeb/Library/CloudStorage/OneDrive-JPL/Documents/GitHub/corgidrp/corgidrp/walker.py", line 75, in walk_corgidrp
    recipes = autogen_recipe(filelist, outputdir, template=template)
  File "/Users/srhildeb/Library/CloudStorage/OneDrive-JPL/Documents/GitHub/corgidrp/corgidrp/walker.py", line 152, in autogen_recipe
    _fill_in_calib_files(step, this_caldb, first_frame)
  File "/Users/srhildeb/Library/CloudStorage/OneDrive-JPL/Documents/GitHub/corgidrp/corgidrp/walker.py", line 193, in _fill_in_calib_files
    best_cal_file = this_caldb.get_calib(ref_frame, calib_dtype)
  File "/Users/srhildeb/Library/CloudStorage/OneDrive-JPL/Documents/GitHub/corgidrp/corgidrp/caldb.py", line 307, in get_calib
    raise ValueError("No valid {0} calibration in caldb located at {1}".format(dtype_label, self.filepath))
ValueError: No valid DetectorNoiseMaps calibration in caldb located at /Users/srhildeb/.corgidrp/corgidrp_caldb.csv`

How can I run the nonlin_e2e test?

maxwellmb, 2024-11-07:
Could it be misidentifying the correct recipe? I don't see DetectorNoiseMaps in either of the two recipes that it should be running.

semaphoreP, 2024-11-07:
Ok let me push a change to obstype-switch

closed 2024-11-07 by hsergi