roman-corgi/corgidrp #251: Kgain: CGI_EXCAM_L1_0000051867_kgain.fits or l1_to_kgain_output/CGI_EXCAM_L1_0000051731_kgain.fits?

hsergi, 2024-11-08:
**Describe the bug**

If I run ``python kgain_e2e.py`` removing the cosmic ray rejection, the run crashes with this message:
```FileNotFoundError: [Errno 2] No such file or directory: 'l1_to_kgain_output/CGI_EXCAM_L1_0000051731_kgain.fits'
```

However when I run it with cosmic ray rejection, it tries to find a different FITS file: ``tests/e2e_tests/l1_to_kgain_output/CGI_EXCAM_L1_0000051867_kgain.fits`` and the e2e test crashes too.

How to fix this? I needed to pass the e2e tests with the sorting algo, which is working

**corgidrp version:**
`` ``cal_sorting``

hsergi, 2024-11-08:
Full error:

`Traceback (most recent call last):
  File "kgain_e2e.py", line 153, in <module>
    test_l1_to_kgain(tvacdata_dir, outputdir)
  File "kgain_e2e.py", line 114, in test_l1_to_kgain
    kgain = data.KGain(kgain_file)
  File "/Users/srhildeb/Library/CloudStorage/OneDrive-JPL/Documents/GitHub/corgidrp/corgidrp/data.py", line 853, in __init__
    super().__init__(data_or_filepath, err=err, pri_hdr=pri_hdr, ext_hdr=ext_hdr, err_hdr=err_hdr)
  File "/Users/srhildeb/Library/CloudStorage/OneDrive-JPL/Documents/GitHub/corgidrp/corgidrp/data.py", line 271, in __init__
    with fits.open(data_or_filepath, ignore_missing_simple=True) as hdulist:
  File "/Users/srhildeb/miniconda3/envs/cgi_etc/lib/python3.7/site-packages/astropy/io/fits/hdu/hdulist.py", line 174, in fitsopen
    lazy_load_hdus, ignore_missing_simple, **kwargs)
  File "/Users/srhildeb/miniconda3/envs/cgi_etc/lib/python3.7/site-packages/astropy/io/fits/hdu/hdulist.py", line 411, in fromfile
    lazy_load_hdus=lazy_load_hdus, **kwargs)
  File "/Users/srhildeb/miniconda3/envs/cgi_etc/lib/python3.7/site-packages/astropy/io/fits/hdu/hdulist.py", line 1064, in _readfrom
    fileobj = _File(fileobj, mode=mode, memmap=memmap, cache=cache)
  File "/Users/srhildeb/miniconda3/envs/cgi_etc/lib/python3.7/site-packages/astropy/utils/decorators.py", line 536, in wrapper
    return function(*args, **kwargs)
  File "/Users/srhildeb/miniconda3/envs/cgi_etc/lib/python3.7/site-packages/astropy/io/fits/file.py", line 170, in __init__
    self._open_filename(fileobj, mode, overwrite)
  File "/Users/srhildeb/miniconda3/envs/cgi_etc/lib/python3.7/site-packages/astropy/io/fits/file.py", line 558, in _open_filename
    self._file = fileobj_open(self.name, IO_FITS_MODES[mode])
  File "/Users/srhildeb/miniconda3/envs/cgi_etc/lib/python3.7/site-packages/astropy/io/fits/util.py", line 390, in fileobj_open
    return open(filename, mode, buffering=0)
FileNotFoundError: [Errno 2] No such file or directory: `'l1_to_kgain_output/CGI_EXCAM_L1_0000051731_kgain.fits'`

semaphoreP, 2024-11-08:
Fixed in PR #247

closed 2024-11-08 by semaphoreP