roman-corgi/corgidrp #267: DQ fields should be integer types not float types

ecady-jpl, 2024-12-09:
**Is your feature request related to a problem? Please describe.**
The definition for the contents of the DQ matrix (section 4.0 of the Corgi-DRP Implementation Document) clearly show it as implemented as a series of bit flags that can be masked to identify bad pixels in general or ones flagged for meeting a certain criterion in particular.  This is fine.  What doesn't make sense is that DQ is implemented in practice with a 64-bit float.  A floating-type should never be used for bit masking as the bit flag positions will change based on the total value of the number.

**Describe the solution you'd like**
Change DQ to an integer type, preferably unsigned (signed has no use case here).

semaphoreP, 2024-12-09:
I'm pretty sure it's an integer right now. Perhaps the documentation is wrong.

semaphoreP, 2024-12-09:
I just spot-checked a couple of files I made with the DRP, and they both are int DQ extensions. I would ask for an example or a bug-reproduction script to see when this is not true.

ecady-jpl, 2024-12-09:
Looks like the documentation is wrong then.  I'll discuss with @mygouf about who needs to do the cleanup on the SSC wiki.  But we can close this up here.

closed 2024-12-09 by ecady-jpl

maxwellmb, 2024-12-09:
@ecady-jpl I'll note that we still think they're 64-bit integers. if I'm understanding the origin of this concern correctly (data volume), you might still want it on your radar.

ecady-jpl, 2024-12-09:
@maxwellmb I am wary of data volume, and do think you ought to scrub it.  But this one is just a straight mismatch--you can't bitmask floats because the bits will float around depending on what bits are set...

maxwellmb, 2024-12-09:
Ok good good. I agree it'd be a problem if it were float. Vanessa also mentioned the data volume thing the other day and the idea of forcing the bit depth for a future release has come up.

agreenbaum, 2024-12-11:
I would also advocate for reducing data volume in cases like this. Happy to join the discussion if/when it comes up.

kjl0025, 2024-12-17:
I just corrected the data format page (https://collaboration.ipac.caltech.edu/display/romancoronagraph/Data+Array+Formats) to say that the DQ is 64-bit integer (as it is currently in the DRP).