roman-corgi/corgidrp #270: Reduce data volume with (optional) forced bit depths

maxwellmb, 2024-12-13:
**Is your feature request related to a problem? Please describe.**
Right now the data volume might get fairly big since we're not forcing any specific datatypes or bit depths and just relying on the 64-bit python default. We'd like to reduce this to lighten the burden on the SSC and future users, by (optionally) forcing the data types to be 32-bit floats for the image and error extensions, and a 16-bit integer for the DQ extension. 

**Describe the solution you'd like**
In order to keep our options open, let's implement this with a drpconfig keyword that toggles us to force the bit-depth described above. We can leave it as True (i.e. forcing bit-depth) by default.

maxwellmb, 2024-12-13:
@agreenbaum, @mygouf, @nasavbailey, @semaphoreP. Please comment if you think anything is missing here.

maxwellmb, 2025-03-17:
Following up on this: we're not dealing with large data volumes right now. Can we punt this to the next release?

nasavbailey, 2025-03-17:
Yes, OK to punt.