roman-corgi/corgidrp #273: Core throughput: CTC 1090881

hsergi, 2024-12-20:
Implement:

1090881 - Given a core throughput dataset consisting of M clean frames (nominally 1024x1024) taken at different FSM positions, the CTC GSW shall estimate the pixel location and core throughput of each PSF.

NOTE: the list of M clean frames may be a subset of the frames collected during core throughput data collection, to allow for the removal of outliers.

hsergi, 2024-12-20:
There's some significant work done towards closing this issue in the ``ct`` branch. On the other hand, the current FDD for CT is going through a review that affects this req. and may change its text. It will be finalized after Jan 10th. I'll likely post the PR review once the new version is accepted. In the mean time, I'll clarify with the customers a couple of questions too.

hsergi, 2025-01-31:
Overview. 

At the end of the collection of the off-axis PSF core throughput data, LSAM, FSAM and FPAM are removed, and the imaging lens is replaced by the pupil lens. One or more pupil images, recommended at least three to five images, are taken to calibrate the total number of counts of the unocculted star. The acquisition of pupil images is part of the CT visit file, which follows the same steps used during TVAC (see below). 

Input data:

Clean frames: A few noiseless OS11 simulations for coronagraphic images(hlc_os11_psfs_oversampled). Plus some pupil images of the unocculted PSF. Units are photoelectrons / pix / second.

NOTE: The values of the FSM positions are not needed in the verification of 1090881. If there is some other application that needs them -not identified at this moment-, they are always available in the L1 Header Keywords FSM_X and FSM_Y.

Output:

Array with the two values of the pixel location and core throughput for each PSF
Use EXCAM pixel (0,0) as the reference point for the locations in this script. Note: The shift from the coronagraphic FPAM mask center during core throughput acquisition and that during coronagraphic observations is the subject of 1090882
Core throughput should be in units of ([0,1]. If there are any headers, add the definition of the core throughput units

hsergi, 2025-01-31:
@maxwellmb can you please check the proposed UTs for 1090881? [Link](https://collaboration.ipac.caltech.edu/display/romancoronagraph/DRP+Test+Definition)

hsergi, 2025-02-03:
I need to include the transmission of the pupil lens relative to the imaging lens when I derive the CT value. The pupil and imaging lenses transmissions have definite measured values that will not change or being measured again. Where can I add those files? I see some filter files under 'corgidrp/data/', which are a similar dataset

hsergi, 2025-02-03:
Although CT is aimed at TTR5 (1F), since corgidrp/data/ has all main bands and sub-bands from 1 to 4. Should I write the pupil vs. direct imaging transmission function accepting any such filter? Otherwise, if it is for only one specific filter, it would be a constant value (e.g., for 1F the ratio is 0.9840324)

semaphoreP, 2025-02-03:
corgidrp/data is a good place to put them. The function should accept any specific filter, but it's ok if only a subset have been implemented for now (if e.g., you don't have all the numbers).

maxwellmb, 2025-02-04:
> [@maxwellmb](https://github.com/maxwellmb) can you please review the proposed UTs for 1090881? [Link](https://collaboration.ipac.caltech.edu/display/romancoronagraph/DRP+Test+Definition)

I've approved these, although when we see the final functions I can't promise we won't want more. It's hard to predict all the unit tests before we see the actual implementation. I'd also imagine that these will come together into a single step function so it'll make sense to bundle these tests with at least 882 and maybe 883 and 884.

hsergi, 2025-02-08:
> I've approved these, although when we see the final functions I can't promise we won't want more. It's hard to predict all the unit tests before we see the actual implementation. I'd also imagine [...]

Comment from @nasavbailey in the review of UT in Confluence ():

"Looking at the code, it seems like the test is to calculate CT on real+simulated images with the same method twice (once by hand with the same steps as in DRP and once by calling DRP directly). If that's the case I think it would always return the same answer? If the method is biased or has a bug, this test wouldn't flag it, since we don't know what the ground truth was. Instead, would it be better to input a simple dummy case for which you know the exact analytic answer? For example, a "pupil" that is a uniform 10x10 square of all 1s and a "PSF" that is a simple small array that you've assigned values by hand? "

Right now the code uses OS11 PSFs and some deterministic pupil images based on TVAC data. The off-axis PSFs do not have a pre-defined location (for the reasons discussed in the resolved comments on Confluence: ) or CT value. As discusses sometime ago in Confluence, these tests are sort of an identity. @maxwellmb: what do you think needs to be done in 1090881?

(see branch ct_109881)

maxwellmb, 2025-02-08:
We need to validate that the code produces the correct value at the output for given inputs. I think it's true that to do this you need to know the hard truth of the CT and star positions of your test images.

hsergi, 2025-02-08:
This truth information is not available if we use OS11 PSFs because it's a full complex e2e simulation where the CT or PSF location is the result of applying the estimation method in the function. The only thing to check in this case is that there are no bugs in the method, which requires someone looking at the code verifying that it follows the definition of CT, and the simple comparison of measured input against output. This was done when porting code from TVAC.

It seems that @maxwellmb and @nasavbailey advocate for using some mock data, with an off-axis PSF that has some pre-stablished values per EXCAM pixel, and similarly for the pupil image, so that one can predict the estimated CT and PSF location. The output should match them.

Is that what you want?

If so, I think we should try to avoid pixelization effects or the UT will be an accuracy test against some pixel resolution and method. That's why I added known at the EXCAM pixel resolution beforehand.

maxwellmb, 2025-02-08:
Well I think it depends on which TVAC port you're talking about. I don't recall clearly each one, but I'd guess some of the DRP tests were comparing against previous TVAC results. 

But yes, you've captured what we're after in general for UT tests. 

My read of the formal requirement is that there's no strict accuracy requirement so pixel level should be enough to pass the requirement. In the end we'll definitely want better, so we'll probably want an issue so we don't forget about it.

maxwellmb, 2025-02-08:
We should get another opinion on that too. 

Also you should see if @juliamilton's function on centroiding helps you do better

hsergi, 2025-02-08:
It's a matter of functionality vs. performance, and I sense different preferences. The idea in the example below applies to any case, not just this UT.

Functional UT: When the result of the step function can be known a priori, the UT can be purely functional. In this case, create an N1xN2 array with some pre-stablished values consistent with the a priori result. It does not even need to be a PSF shape. Test passes if the measured properties are the same as the pre-stablished ones. Repeat for different choices of values as needed. Outcome: The function works as expected. If realistic inputs are more complex than the info in the N1xN2 array (e.g. PSF are continuous), little or nothing is known about the function's performance. Performance is part of a trade study outside the scope of the UT (for instance, it could be, to some extent, part of the e2e test).

Performance UT: Create an (mN1)x(mN2) array, with m>1, from some pre-stablished properties (center, CT, etc). Bin the array into the N1xN2 size and measure the properties with the step function. Alternatively, sample an analytical expression at the N1xN2 pixels of the array. The test passes if the properties are recovered within some tolerance. Outcome: The function works as expected within the tolerance though it may not function (one could add a bug that generates random changes of any property within some fraction of the tolerance or anything wrong below the tolerance). The functionality is verified by someone's visual inspection of the code making sure there are no bugs.

Notice that OS11 PSFs or TVAC data have no pre-stablished properties on-detector. They are useful for e2e tests that include noise with functions that have passed the UTs above, but not as much for UTs themselves.

maxwellmb, 2025-02-08:
I've got to admit I don't fully understand the distinction in the way you've described things. All the code should both work (be functional) and have reasonable performance (admittedly in some cases we may need to define this still).  The tests should demonstrate that for inputs that are reasonably close to the expected real inputs we can recover the expected outputs. If you want to inject multiple input datasets to demonstrate this, then go for it! In this case it seems like CT and position are some of the more important measurements, so those are the ones we need to test.

nasavbailey, 2025-02-10:
I'm also not sure I follow the 'performance' definition proposed. I do agree that units tests must demonstrate that the function produces the right answer for known cases (what I think is called 'functional' above) and also that it will operate on realistic data (produce an answer, even if the ground truth is not known).

hsergi, 2025-02-10:
I've been troubleshooting my laptop all day. And need to go to Mantech in person to get my JPL laptop back.

I'm adding this with the hope you can review it and I can move on later today. It's a practical proposal after what I heard in the meeting this morning in other cases.

hsergi, 2025-02-10:
v3.0.2: functionality.

IMO, functionality must include 6-9 l/D for v3.0.1. **If you think it's not for
v3.0.2, let me know**. If so, do we need to consider the effects of obscuration
near the OWA? I think we do. Do we need to consider the effects of PSF distortion
near the IWA? I do not think so.

The distortion effects near 3 l/D can be ignored because the closest PSF has
its center at 6 l/D and the PSF intensity at r=3 FWHM (l/D) is 2**(-4*(3)^2)
~1.5e-11 that of the PSF's peak for a Gaussian profile, .

The OWA is at 9.7 l/D. Then, for sources within 6-9 l/D, the PSF's peak is not
obscurated. It falls within the HLC area. 

CT is defined in an area of pixels whose value is >= 50% PSF's peak. Since
FWHM=50% PSF's peak, the CT area is PSF's peak + circle of radius 0.5*FWHM 

For an Airy disk, FWHM~1l/D, and the CT area is PSF's peak + circile of radius 0.5 l/D.
Therefore, the farthest distance from mask's center is 9.5 l/D. Obscuration starts
at 9.7 l/D.

EXCAM pixels are 21.8 mas or 0.44 l/D. 9.5 l/D is 21.6 +/- 0.5 pixels,
depending on where the FPM's center falls within an EXCAM pixel. And
9.7 l/D is 22.04 +/- 0.5 pixels. Thus, there's EXCAM pixels within the CT area
that are affected by obscuration. For instance, if the FPM's mask falls close to
the edge of an EXCAM pixel.

Moreover, if the method to estimate the PSF's center is larger than 1 pixel
(1 pixel=0.44 l/D), then the PSF's center will also be affected by obscuration
for sources close to 9 l/D.

The centroid method probably uses a minimum of 3x3 pixels, making it worse.

That's why I chose the maximum to avoid the obscuration issue.

Obscuration can be simulated to a some precision as zeroing the signal outside
a circle with radius 9.7 l/D.

Therefore, I propose three ways to test this UT. Please choose one or more. I'm fine implementing more than one if needs be.

hsergi, 2025-02-10:
**1/** We simulate a 2D Gaussian with FWHM=1 l/D at different locations
between 3-9 l/D taking into account obscuration or not. These PSFs need
to be simulated with smaller pixels. The PSF's true center is well known
within one small pixel. The CT needs to be estimated.

For the pupil images, it could be a couple of simulated pupil images with two illumination values, because in this way, we test that the step function splits correctly >1 puil images and averages them.

The test would be to compare the retrieved values with the input ones.

Using a centroid, taking into account the obscured area will give better results than the maximum, though this can be postponed as an Issue that may be an extension of the centroid function written by Julia. It's not a simple centroid -it needs to add the obscuration. I tried that quickly using "center_of_mass" from the scipy library and PSF's centers are biased, as expected, w/o accounting for the effect of the OWA.

We will retrieve an effective value of how much disagreement is b/w simulated PSF values (at high resolution) and the actual ones, and we can use it for setting a tolerance on the unit test. 

Since there's no a priori knowledge on how much the disagreement will be, one still needs to look at the code and make sure it has no bugs.

**2/** Similar as before, using OS11 simulations. The expected CT value though is not known a priori. OS11 simulated off-axis PSFs are 5x oversampled. I could derive the PSF's center and CT in the oversampled set and compare with the values retrieved by the step function (that uses EXCAM pixels). 

Again there's no a priori known tolerance. It will be a small number known after looking at the differences.

**3/** Write a synthetic data set at EXCAM pixel pitch. One knows where the maximum is and how much is the CT exactly. One can vary the values and location for having more diversity. The tolerance here is 0. The drawback is that we do not know how close to an actual PSF ("much greater sampling") the results will be.

hsergi, 2025-02-10:
@maxwellmb and @nasavbailey : do you have any preference on using 1, 2, and/or 3 for the UTs of 1090881? Open to suggestions

maxwellmb, 2025-02-11:
I don't fully understand what you're proposing for /3 since there aren't many details about the EXCAM pixel pitch. 

I'm leaning towards 1/ since you can validate the CT and x,y positions. I appreciate your point that it doesn't take care of the PSFs near the edge of the field and this is something we can work on validating/improving as our simulation infrastructure improves. 

Even for 1, we can set reasonable tests on the performance of the algorithm. If we're working with perfect Gaussians our ability (especially in a high SNR case) to recover the CT and positions should be really quite good. I think a low bar to pass would be knowing the position to 1/2 lambda/D and the core throughput to within 5-10%.

nasavbailey, 2025-02-11:
Ah, I think I see part of the issue with the position part of this req't. The requirement is to "...estimate the pixel location and core throughput of each PSF," not to estimate the true on-sky (distortion-corrected) position of the PSF. That language was chosen to avoid the issue you're raising, Sergi, that neither the center of light position nor the peak pixel is not the true star position near the IWA and OWA. So no position correction for obscuration needed for the pixel location for this req't! I would personally advocate for estimating the PSF pixel location using the center of light in a fixed aperture 1l/D diameter, rather than taking the brightest pixel location, but it's not the hill I'm going to die on.

If 3) is saying to create toy model pupils and PSFs where their properties are exactly known ahead of time, then I agree this is a good test. The pupil doesn't have to look like CGI's and the PSFs could be 2 simple pyramids centered at and between pixels. as you say, tolerance on flux and location is 0 (or w/in some tiny tolerance). 

1) The fractional flux of a Gaussian contained within its FWHM is analytically known, as is its centroid, so I agree this is a good way to test how accurately the step function recovers it, given the finite image sampling. I don't understand "These PSFs need to be simulated with smaller pixels," though. I would think this data should be simulated at the CGI plate scale with FWHM = 1 l/D?

hsergi, 2025-02-11:
I think it's fine to do two UTs: 1/ w/o obscuration and 3/. Agreed on everything above. I wanted to share it before converging.

@nasavbailey : The PSFs will be sampled at EXCAM pixel resolution. For 1/, since the "true" PSF may fall within the pixel, there is the commonly used procedure to start with smaller pixels, plug their centers into the analytical expression of the PSF shape, and, then, bin them to EXCAM's pixel pitch, using some standard method. This is a simple way to simulate the cumulative light that falls on an EXCAM's pixel. Alternatively, one can associate to every EXCAM pixel the integrated value of the analytical expression over one EXCAM pixel. The small pixel approach is simply a Riemann sum of the latter (direct sum of each small volume).

Tbh, I don't mind which method. I may use integrate from scipy. The other one is a DIY

maxwellmb, 2025-02-11:
I think you could use an astropy 2D Gaussian model. It should do the oversampling thing you're talking about automatically.

hsergi, 2025-02-11:
If you mean this one: [here](https://docs.astropy.org/en/stable/api/astropy.modeling.functional_models.Gaussian2D.html), I think it evaluates the analytical expression at the pixel's index value, w/o integrating over a pixel size. Maybe you know of another one?

hsergi, 2025-02-11:
This is the summary (in 1/ we deal with subpixel and some small tolerance. I think we can consider exact agreement in the other test, by using the pixel's center only. Still basic functionality is tested, besides non-trivial CT.

Two sets of UTs:
All are simulated frames at EXCAM pixel pitch
1/ Simulate 2D Gaussian with FWHM=1 l/D at different locations within a pixel, and a couple of pupil images
2/ Define some synthetic shapes on EXCAM whose center and CT is well-know in advance
Both tests:

- Check that the estimated output location is the same as the input one, within some small tolerance, for 1/ and exact for 2/.
- Check that the estimated output core throughput value is the same as the input one, within some small tolerance, for 1/ and exact for 2/.

Let me know if this is OK. I'll update Confluence

maxwellmb, 2025-02-11:
It's not clear to me what the difference is between 1/ and 2/

hsergi, 2025-02-11:
2/ As Vanessa says, the PSF can be a pyramid. The center location and the CT can be known to infinite accuracy. That's not the case in 1/.

maxwellmb, 2025-02-11:
Doesn't 1/ serve this purpose anyway?

hsergi, 2025-02-11:
2/ proves that the step function does exactly what is meant for (purely functional). No astronomy. 1/ shows how it works for a reasonable PSF shape. Many would say 1/ is enough. 2/ is more of a software engineer perspective

maxwellmb, 2025-02-11:
1/ would be enough for me. We don't need it to work for arbitrary PSFs.
Just the ones that matter.

hsergi, 2025-02-11:
@nasavbailey : Is 1/ enough? Or would you like to add 2 too? To be clear, 2/ is setting on an EXCAM frame some values (no subpixel) with known CT and location. Conceptually it is independent of any modeling/bug biases that may occur within the tolerance of 1/. Max does not see it necessary. Since 1/ deals with subpixel with a 2D gaussian, we probably do not need to deal with other shapes at subpixel scale, like the pyramid.

I'm fine either way since this is meant to work with PSFs, 1/ is enough. 2/ is IMO good practice

hsergi, 2025-02-11:
> If you mean this one: [here](https://docs.astropy.org/en/stable/api/astropy.modeling.functional_models.Gaussian2D.html), I think it evaluates the analytical expression at the pixel's index value, w/o integrating over a pixel size. Maybe you know of another one?

Can you check on this? Otherwise, I'd use integrate.

hsergi, 2025-02-12:
Yes.

FYI, the centroid test, which is simply a sum and multiply in a box with +/- 1 l/D agrees within 0.01 pixel for any random location within an EXCAM pixel. That is, assert np.all(np.abs(diff_psf_loc) <= 0.01) passes.

This is the few lines. As far as obscuration is not considered is straightforward:

```
pix_max = np.unravel_index(psf.data.argmax(), psf.data.shape)
            # Cut out PSF within +/- 3 pixels (+/- 1.3 l/D for band 1F)
            psf_cent = psf.data[pix_max[0]-3:pix_max[0]+4, pix_max[1]-3:pix_max[1]+4]
            y, x = np.indices(psf_cent.shape)
            y_cent = (y*psf_cent).sum()/psf_cent.sum()
            x_cent = (x*psf_cent).sum()/psf_cent.sum()
            # Add centroid shift relative to maximum location
            psf_pix += [[pix_max[0]+y_cent-3,pix_max[1]+x_cent-3]]
```

I've reached out to Julia earlier today and shared it. Also, A.J. ElDorado Riggs is asking me to use the TVAC function psffit instead. I have forward his request to Vanessa.

maxwellmb, 2025-02-13:
> After an email exchange with AJ and Vanessa: we are good considering the
> centroid option vs psffit for v3.0.2. While I wait to hear from Julia, I'll
> move on with this and other UTs and reqts.

Feel free to just peek at her branch that has already been PRed. You should
find the function relatively easily. No need to wait for a response.

juliamilton, 2025-02-13:
@hsergi updated i put my updated centroiding function (centroid_with_roi) in astrom.py: https://github.com/roman-corgi/corgidrp/blob/nd_filter/corgidrp/astrom.py

hsergi, 2025-02-22:
PR submitted: [link](https://github.com/roman-corgi/corgidrp/pull/305)

closed 2025-03-15 by hsergi