roman-corgi/corgidrp #274: Core throughout: CTC 1090882

hsergi, 2024-12-20:
Implement:

1090882 - Given 1) the location of the center of the FPM coronagraphic mask in EXCAM pixels during the coronagraphic observing sequence and 2) the FPAM and FSAM encoder positions during both the coronagraphic and core throughput observing sequences, the CTC GSW shall compute the center of the FPM coronagraphic mask during the core throughput observing sequence.

hsergi, 2024-12-20:
Note: Use conversion factors from um to mas for FPAM, FSAM and EXCAM (Confirm values from TVAC and any plans to remeasure them in flight)

hsergi, 2025-01-31:
There are no plans to measure them during flight. Overview:

The FPAM's center will be used as the center of the CT correction (see 1090883).

For the DRP release 3.0.2 (April 1st. 2025), there is not an obvious application of the FSAM center within the CT correction in the TTR5 area. However, since it could be needed in some future analysis of sources near the OWA, we interpret 1090882 as computing the FSAM and FPAM centers independently, from their delta values and their rotation matrices.

Input data:

Center of the FPM coronagraphic mask in EXCAM pixels during the coronagraphic observing sequence:

Two arbitrary EXCAM pixels with some margin to include the HLC mask
FPAM/FSAM positions

Two different FSAM/FPAM positions during both the coronagraphic and core throughput observing sequence. FSAM/FPAM positions: FSAM_H/FPAM_H and FSAM_V/FPAM_V. Units are um (microns)
Rotation matrices

Files: fpam_to_excam_modelbased.fits and fsam_to_excam_modelbased.fits, found in [Alfresco (JPL Firewall)](https://alfresco.jpl.nasa.gov/share/page/site/cgi/documentlibrary#filter=path%7C%2FRoman%2520CGI%2520Collaboration%2520Area%2F04%2520-%2520Coronagraph%2F15%2520-%2520CGI%2520II%2526T%2FInstrument%2520Integration%2520and%2520Test%2FCGI%2520I%2526T%2520Phase%2520CD%2FII%2526T%2520Calibration%2520Products%2FTransform%2520Matrices%7C&page=1)
Store rotation matrices in a similar way to DetectorParams (see caldb.py)
Python algorithm: 
delta_pam = np.array([[dh], [dv]]) # fill these in

M = np.array([[ 0. , value1], [value2, -0. ]], dtype=float32)

delta_pix = M @ delta_pam

Output:

The output is two EXCAM pixels indicating the center of the FPM coronagraphic mask during the core throughput observing sequence wrt EXCAM (0,0) pixel

hsergi, 2025-02-04:
@maxwellmb Can you please review the UTs of 1090882? [Link](https://collaboration.ipac.caltech.edu/display/romancoronagraph/DRP+Test+Definition) I've taken into consideration your comments about the UTs for 1090881 and the fact that we may bundle the reqts when submitting a PR

maxwellmb, 2025-02-05:
> Max asked me to store rotation matrices (2x2 numbers) in a similar way to DetectorParams. Does it mean to add a new key in default_values? ([@maxwellmb](https://github.com/maxwellmb) )

I meant to create a new calibration file type that is initialized in a similar way as DetectorParams. See for example lines 349 to 353 in `caldb.py`.

hsergi, 2025-02-05:
That creates a FITS file with array([[0.]], dtype='>f8'). 

I reviewed caldb.py, data.py and DetectorParams and I'm not sure what to do with the two rotation matrices. Sometimes, calibration files belong to calibration classes (nonlinearity, kgain, ...). Do I need to create a class for the rotation matrices?

maxwellmb, 2025-02-05:
Yes, we need a new type of calibration file for this. And you'd pre-populate it in caldb.py with the tvac rotation matrices.

maxwellmb, 2025-02-05:
> [@maxwellmb](https://github.com/maxwellmb) Can you please review the UTs of 1090882? [Link](https://collaboration.ipac.caltech.edu/display/romancoronagraph/DRP+Test+Definition) I've taken into [...]

Ok, I reviewed the 1090882 UTs. It looks fine to me with the same caveat that I'll really have to re-evaluate things when I see the implementation.

hsergi, 2025-02-05:
> Yes, we need a new type of calibration file for this. And you'd pre-populate it in caldb.py with the tvac rotation matrices.

Ty. Just to make sure: calibration **file**, not class. Right?

maxwellmb, 2025-02-05:
Sorry that was unclear. Yes I mean it'll need a new class

maxwellmb, 2025-02-05:
probably one calibration type for both matrices is fine

hsergi, 2025-02-08:
> Ok, I reviewed the 1090882 UTs. It looks fine to me with the same caveat that I'll really have to re-evaluate things when I see the implementation.

@maxwellmb . Some comments from @nasavbailey  on the UT review: [1090882](https://collaboration.ipac.caltech.edu/display/romancoronagraph/DRP+Test+Definition)

"Test with a mock rotation translation/matrix that is easily analytically knowable and compare to that. eg: x scale is 10% larger than y, test a motion. Rotate by 30dgr. B/c I think the point is to test that the trans/rot matrix is loaded and processed correctly, not to test whether it matches the version used for TVAC."

So far, I have understood that we would test the functionality of recovering an FPM center with the TVAC matrices. Do we also need to test the method for a transformation defined in the test that is different than the TVAC one?

maxwellmb, 2025-02-08:
We just need the test to show that the output is what was expected I think. That whatever loaded in did so correctly.  It could be TVAC or an arbitrary other one. We probably want an asymmetric matrix so that there's no ambiguity when comparing input vs. output (similar to @nasavbailey's) online comment.

hsergi, 2025-02-10:
> Yes, we need a new type of calibration file for this. And you'd pre-populate it in caldb.py with the tvac rotation matrices.

This has been implemented. It can be reviewed in the branch ct_1090882 or during the PR

hsergi, 2025-02-10:
@maxwellmb and @nasavbailey : Regarding the UTs for this one -which will be bundled with 1090881 and 1090884:

Can we assume that the validation of the rotation matrices was developed and checked by the IIT team during FFT and TVAC CGI tests? In this way, this requirement is not a development but a porting task: the unit test of the step function would check that the values of the rotation matrices in the calibration file agree exactly with those from the TVAC files (from Alfredo). Furthermore, the IIT team shared and checked the python operation that needs to be done to get the results. 

The unit test would copy that, derive the changes and compare with what the function outputs for one or more non-trivial FPM centers and delta H/V values. The results must agree in all cases.

Is that acceptable? Open to suggestions, though I'd like to emphasize the possibility of avoiding a full review of PAM vs. rotation matrices that has been done by JPL CGI team in the past and validated by them. We can ask (again) for documentation as I did back in December

maxwellmb, 2025-02-10:
> > Yes, we need a new type of calibration file for this. And you'd pre-populate it in caldb.py with the tvac rotation matrices.
> 
> This has been implemented. It can be reviewed in the branch ct_1090882 or during the PR

FYI, I'll probably wait until the PR to review it so that I can see it all in contact.

maxwellmb, 2025-02-10:
> [@maxwellmb](https://github.com/maxwellmb) and [@nasavbailey](https://github.com/nasavbailey) : Regarding the UTs for this one -which will be bundled with 1090881 and 1090884:
>
> Can we assume that [...]

Yes, this is what I was thinking. Let's assume that those values are correct and the unit test just needs to make sure that the calibration product, as constructed and read-in ,reflects those values. 

> The unit test would copy that, derive the changes and compare with what the function outputs for one or more non-trivial FPM centers and delta H/V values. The results must agree in all cases.
>
> Is [...]

This sounds good to me.

hsergi, 2025-02-11:
Sounds good to me. FYI, I'm flying tomorrow to Spain for 3 weeks. I'll be working remotely though I'd like to get all this done asap and submit PRs.

hsergi, 2025-02-13:
I added the TVAC files fpam_to_excam_modelbased.fits and fsam_to_excam_modelbased.fits using git lfs track to test_data, and MANIFEST.in. These two files are only needed for the UTs of 1090882. There's now a calibration file with the values of these two transformation matrices in the core throughput branch. 1090882 will be PRed together with 1090881 and 1090884

hsergi, 2025-02-22:
PR submitted: [link](https://github.com/roman-corgi/corgidrp/pull/305)

hsergi, 2025-03-15:
PR closed and merged to main: https://github.com/roman-corgi/corgidrp/pull/305

closed 2025-03-15 by hsergi