roman-corgi/corgidrp #275: Core throughput: CTC 1090883

hsergi, 2024-12-20:
1090883 - Given 1) an array of PSF pixel locations and 2) the location of the center of the FPAM coronagraphic mask in EXCAM pixels during core throughput calibrations, and 3) corresponding core throughputs for each PSF, the CTC GSW shall compute a 2D floating-point interpolated core throughput map.

hsergi, 2025-01-31:
By default, 1090883 will consider a square grid of (x,y) values for the core throughput map. The grid size will be an argument to the function. It will also allow the user to parse an optional, pre-defined set of points.

**Input data**

- Core throughput cal file providing:

> - The center of the FPAM mask (two values) with respect to EXCAM's pixel (0,0) during core throughput observations 
> - Nx2-dimensional array PSF pixel locations (two values per PSF) with respect to pixel (0,0) within the area of the HLC mask
> - Nx1-dimensional array of core throughput values between (0,1]. One value for each PSF

- Target pixels for the core throughput map:

> Default option: square grid defined by x_range = [xmin,xmax], y_range = [ymin,ymax], n_gridx = number_of_x_gridpoints, n_gridy = number_of_y_gridpoints
> - Alternatively, allow to parse a user-defined list of positions

**Method**

- Use bi-linear 2D interpolation method and scipy interpolation method griddata for sparse data

**Output**

- A core throughput map. That is, a 3-dimensional array (x,y,ct) where (x,y) is the position of each target pixel location wrt the FPM center, and ct  is the interpolated core throughput value at each location
- Return the core throughput map only at valid locations. That is, if some location(s) lies outside the input CT dataset, remove it from the output
- If all locations fall outside the area defined by the input CT dataset, raise Exception.
- Add this rule in the function docstring

hsergi, 2025-02-10:
This function uses well-known python libraries. We use an interpolation method from scipy (interpolated.griddata). If we assume that this interpolation method has been validated, the unit tests could be:

1/ First, use OS11 simulations to estimate a PSF at one of its locations, using the N-1 OS11 PSFs. Compare the CT of the OS11 PSF with its interpolated result. Repeat for a few locations b/w 6-9 l/D. The (expected small) tolerance will be dictated by the results. Again, one will have to make sure that there are no other minor bugs looking at the code's implementation and scipy documentation for this interpolation function.

2/ Second, verify that if there's some invalid positions for the interpolation, there are no results outputted for them. The output is valid locations with their interpolated value. Lastly, if there's no valid position at all, an exception is raised.

Is that a good set of UTs @maxwellmb @nasavbailey ?

hsergi, 2025-02-26:
It seems we are gearing to leave OS11 PSFs for an e2e test, and consider the known 2D PSFs. This is the latest suggestion on Confluence (DRP test definition): "First, use a set of 2-D Gaussian PSFs with known locations and CT (CT has a linear, radial profile): This is the reference set. Derive the estimated CT at some other locations, including a grid of points and a user-provided grid of points. Show that the results match exactly, or with some small TBD tolerance. " Note: The baseline interpolation method is "linear". Thus, the need for a linear CT profile if one aims at matching results of new positions and interpolated results exactly. However, "linear" interpolation in 2D sometimes involves bi-linear interpolation, and in that case it may not match exactly, though it will within a small tolerance error. I have to write it and check it

hsergi, 2025-02-26:
Regarding 2/, adding a clarification of which locations are invalid (Vanessa and I agree that this is a better way to describe it). DRP test definition in Confluence updated

Second, verify that if there's some invalid positions (*) for the interpolation, there are no results outputted for them. The output is valid locations with their interpolated value. Lastly, if there's no valid position at all, an exception is raised.  (*) Invalid positions are defined as:  r<r_min, r>r_max, and for cases where only part of the FOV was sampled, trying to interpolate to a location that is outside the "convex hull". By default, r_min is 3 lam/D and r_max is 9.7 lam/D, although these are keyword arguments that the user may change.

hsergi, 2025-03-12:
> @hsergi I wrote some interpolation code and tests. Please see [here](https://collaboration.ipac.caltech.edu/display/romancoronagraph/Max%27s+Interpolation+idea).  

 _Originally posted by @maxwellmb in [#6](https://github.com/roman-corgi/corgidrp/issues/6#issuecomment-2707958366)_

hsergi, 2025-03-12:
@maxwellmb: @nasavbailey is fine with your proposal for this delivery (although she has not been able to review the implementation in detail). It's fine to have the test data that you suggest and the method, and tests. She suggests to add an explicit test for  "to confirm that the angular mod produces the expected results (eg: sample at 0, 60; requesting 80 will produce the interpolated value for 20)", which I think your implementation does already.

I'd write some of the limitations of the tests and methods in the Doc string. Also, the structure of the interpolation function includes some class objects, as agreed last week, and I'd be adjusting your codes to comply with that.

This would be the basis for the CT map generation. I may have a couple of technical comments when the input locations where the CT is interpolated is an array and not a single location once I review the azimuth conditions imposed on the target location.

maxwellmb, 2025-03-12:
Sounds good to me. 

From my perspective you have what you need to round this out now, right?

hsergi, 2025-03-19:
PR submitted, comments applied (add to the UT test a different location of the the CT and Cor FPM's centers), approved and merged into main: https://github.com/roman-corgi/corgidrp/pull/331

semaphoreP, 2025-03-24:
@hsergi The CT map has been added, so do you think we can close this issue?

hsergi, 2025-03-24:
CT map PR closed: https://github.com/roman-corgi/corgidrp/pull/346

closed 2025-03-24 by hsergi