roman-corgi/corgidrp #276: Core throughput: CTC 1090884

hsergi, 2024-12-20:
1090884 - Given 1) a core throughput dataset consisting of a set of clean frames (nominally 1024x1024) taken at different FSM positions, and 2) a list of N (x, y) coordinates, in units of EXCAM pixels, which fall within the area covered by the core throughput dataset, the CTC GSW shall produce a 1024x1024xN cube of PSF images best centered at each set of coordinates.

hsergi, 2025-01-31:
**Input**: N are the same PSFs as in [1090881](https://github.com/roman-corgi/corgidrp/issues/273). That is, the calibration CT dataset.
(x,y) are coordinates relative to the FPAM mask's center.

**Method**: Use the output of 1090881 and 1090882 to derive relative PSF position's with respect the FPAM mask's center

**Output**: CoreThroughput calibration file

hsergi, 2025-02-04:
@maxwellmb @semaphoreP : I'm going to write a new class in data.py: CoreThroughputCalibration, CTCalibration or similar (any preference?) and use this issue to given a CT dataset, call the functions in 1090881 (PSF's centers and CT values) and 1090882 (FPM masks centers) to create a calibration file. 

Let me know if this is fine to resolve this issue. I'll probably PR 1090881-2-3-4 bundled together.

maxwellmb, 2025-02-05:
Ultimately I think you just want a single step function that accepts the input Core throughput dataset and then creates the CoreThroughputCalibration (slight preference for this name since its more descriptive). I think creating the calibration itself would satisfy 1,2 and 4. 

We'd then need a separate function that could satisfy 1090883.

hsergi, 2025-02-05:
Yes, agreed. An example or advise on my technical question about storing data with different structure in the same calibration file would be helpful. I've only worked or seen same structure: single scalars or data that fit a single numpy array (although the dimensions may store different parameters)

maxwellmb, 2025-02-06:
Since the arrays are different sizes I think you might just have to make an extra extension that some of the data gets stored in.

hsergi, 2025-02-06:
That works with FITS. I'm asking specifically about the structure in the call of ``super().__init__(data_or_filepath, ...`` all data I've seen can be put into one single array. Not sure how to handle "N-PSF frames, Nx3 array with positions and CT values, and FPAM/FSAM centers during coronagraphic and core throughput observing sequences (total of 4 positions, 2 EXCAM pixels each). " The PSF frames size is variable (N1 set by a keyword argument). We do not use dictionaries for passing data to calibration classes?

maxwellmb, 2025-02-06:
I suggest looking over some of the other types of calibration products, it looks like there are a few different examples of different ways to construct a calibration file. The KGain and DetectorParams objects both contain non-standard data, but in different ways. There might be more, but they are the ones that just popped out when I looked at it now.

hsergi, 2025-02-10:
The cal file should not be a problem after having written the ones for the rotation matrices.

For the UT, remember this reqt has been interpreted as generating the CT cal file only. Nothing else.

The UT would be to check that the calibration file agrees with the input data that are used to call the function.

Would that be acceptable @maxwellmb @nasavbailey ?

maxwellmb, 2025-02-10:
I guess the UT for the call file will want to check the CT and the positions (from 881). I suppose we'll want to check that the data was put into the right place too (for the actual images that we're saving). 

As I said in the UT Table, until I see the full implementation code I can't promise that I won't ask for other unit tests, but this definitely sounds like a UT that we'll need to do.

hsergi, 2025-02-11:
I'm calling the attention that 1090884 should be a very straightforward UT given that 1090881 and 1090882 have their UTs, and these three reqts. will be bundled together in a PR. P.S. 1090883 is an application of the CT cal file to find interpolated values. I'll PR it separately since it requires to write a helper function and it makes more sense when the CT Calibration calss has been reviewed. P.P.S. The method is coded up, not the helper function.

For now, I'll update the Confluence tables.

maxwellmb, 2025-02-11:
Great points all around. Works for me. Safe travels!

hsergi, 2025-02-22:
PR submitted: [link](https://github.com/roman-corgi/corgidrp/pull/305)

hsergi, 2025-03-15:
PR closed and merged to main: https://github.com/roman-corgi/corgidrp/pull/305

closed 2025-03-15 by hsergi