roman-corgi/corgidrp #285: Crop Function Upgrade: Handle modes other than HLC

ell-bogat, 2025-01-14:
**Is your feature request related to a problem? Please describe.**
The crop function in l3_to_l4.py currently assumes the input data is taken using the HLC mode, and raises an error if not. 

**Describe the solution you'd like**
Expand the crop function to handle additional instrument modes, and adjust the default behavior of the crop function based on the mode. E.g. for the HLC mode, the crop function assumes by default that you want to crop to a region the size of the HLC dark hole, centered on the star position, so we should adjust the size of the crop region based on the dark hole size for a given mode. The header keywords for the star position would probably remain the same, but I'm not certain yet.

ell-bogat, 2025-01-27:
Table 1 of this paper describes a bunch of possible modes and the associated Precision Alignment Mechanism (PAM) positions associated with them: https://arxiv.org/pdf/2108.05986

And this page in the Confluence space outlines all the header keywords associated with the PAM: https://collaboration.ipac.caltech.edu/display/romancoronagraph/CFAMNAME%2C+DPAMNAME%2C+FPAMNAME%2C+FSAMNAME%2C+LSAMNAME%2C+SPAMNAME

maxwellmb, 2025-12-17:
@everlastingEric did we finish this?

everlastingEric, 2025-12-18:
I think I addressed the issue here but if I remember correctly @ell-bogat was also working on adding new features to the crop function, @ell-bogat did you ever address this in a PR?

ell-bogat, 2025-12-18:
Thanks @everlastingEric , this issue definitely looks closed to me! I think there are some additional updates I had in mind, namely making sure that when we crop from odd to even (or vice versa) image sizes that the half-pixel shift in any saved coordinates in the header is updated correctly.