roman-corgi/corgidrp #286: Incoming DESMEAR flag not taken into account in desmear step function

kjl0025, 2025-01-14:
**Describe the bug**
The desmear step function does not check the DESMEAR header value to determine whether it should desmear, but the requirements say that the desmearing should happen according to the flag value.

**To Reproduce**
Run the step function on a dataset with at least some frames with 'DESMEAR' = False.

**Expected behavior**
If one runs the step function on a dataset with at least some frames with 'DESMEAR' = False, those frames will still get desmeared.

**corgidrp version:**
v1

**Additional context**
I think it should desmear according to the DESMEAR header value, and maybe an optional input could be added to allow the user to override that determination.

maxwellmb, 2025-12-17:
@kjl0025 is this still relevant? When would a desmear header keyword like this be generated?

semaphoreP, 2026-01-06:
I think the question is that there are two possible uses for the DESMEAR keyword:
  1. DESMEAR indicates if the desmear operation has already been applied by the pipeline
  2. DESMEAR indicates if the pipeline should apply the desmear operation or not.

Right now, DESMEAR behaves as option 1, which makes sense to me. Kevin is saying that in some parts of the FDD, we have the wording that "Apply desmear algorithm if desmear flag is True", which implies that DESMEAR should be the second behavior. In the second case, who is in charge of setting DESMEAR to determine if we should be applying it (another DRP pipeline step?).

maxwellmb, 2026-01-07:
Thanks for catching that Kevin!