roman-corgi/corgidrp #291: Need to double check the PAM Name keywords vs. their H/V positions

maxwellmb, 2025-01-29:
**Is your feature request related to a problem? Please describe.**
The PAM name Keywords CFAMNAME, DPAMNAME, FPMAMNAME, LSAMNAME, SPAMNAME will be set by the SSC to the closest "Enum" value based on the closest match to the H and V values in the Tables on [this](https://collaboration.ipac.caltech.edu/display/romancoronagraph/CFAMNAME%2C+DPAMNAME%2C+FPAMNAME%2C+FSAMNAME%2C+LSAMNAME%2C+SPAMNAME) webpage. This means that the mechanism might be incorrectly position, but still give a valid PAMNAME keyword. 

**Describe the solution you'd like**
We want a check early on the in DRP that checks that the PAMS are within [TBD] units of the desired H and V positions. It would then write the correct Enum to either a new keyword or overwrite the previous value (while still saving the original somehow). We might want some kind of step function that is called "update_headers", where we'd put other similar types of checks or adjustments. 

**Additional context**
Add any other context or screenshots about the feature request here.

nasavbailey, 2025-01-30:
This issue came up today during a core throughput implementation telecon. CT only needs to sort FPM vs. pupil plane images, and we decided that the L1 header values (the closest named position) is sufficient for that application. 

My reading of this is "We should be checking to see if a PAM command failed and left us in a weird state."  I'd say we should rely on CGI's command error checking to abort if a command fails. And, on the other hand, there are cases like spectroscopy where we intentionally move the slit away from its home position, so this check would flag those as bad.

I suggest we put this on the 'low priority' list until we have a specific need.

maxwellmb, 2025-01-30:
Sounds good. Your read was correct. I've adjusted it to low priority and removed it from the R3.0.2 milestone