roman-corgi/corgidrp #307: Helper function to transform distortion params into a 2-D map

manduhmia, 2025-02-25:
doesn't have to be this PR, but these 20 lines (taking an array of distortion params, and turning it into a 2-D distortion map) would probably be useful to have as a separate function, since I imagine you might be calling it in multiple places.

_Originally posted by @semaphoreP in https://github.com/roman-corgi/corgidrp/pull/284#discussion_r1946879861_

maxwellmb, 2025-03-12:
@semaphoreP, do we need this for R3.0.2? The FDD makes it seem like they want a "map" and if we're not going to provide a map, we should ask for an ECR.

semaphoreP, 2025-05-06:
@manduhmia, is this issue done with PR #410 , or still needs work?

manduhmia, 2025-05-09:
@semaphoreP yes, this is closed with PR #410

closed 2025-05-09 by semaphoreP