roman-corgi/corgidrp #325: Calculating a new ND OD value should take an FPamFSam cal file

maxwellmb, 2025-03-07:
**Describe the bug**
Calculating a new ND OD value should take an FPamFSam cal file, but the FPamFSam file PR isn't fully merged yet. Once it is, then we need to have the ND function that uses it accept it. 

**Expected behavior**
calculate_od_at_new_location should take an FPamFSam cal file.

hsergi, 2025-08-15:
@semaphoreP Should I work on a branch based on what is in main or develop for this issue? PR to develop?

semaphoreP, 2025-08-15:
develop for both.

hsergi, 2025-08-19:
The variable ``transformation_matrix_file`` is only present in nd_filter_calibration.py and test_nd_filter_calibration.py. FpamFsamCal files are used in other functions by passing an instance of the calibration class. See for instance, CT functions. It makes sense to use a similar approach across the pipeline. Hence, I have replaced ``transformation_matrix_file`` by an instance of FpamFsamCal. 

One step that was needed in the test function is to tune the values of the FPAM displacement in H/V so that the total displacement in EXCAM pixels is such that allows one to forecast the result of the bilinear (default method) interpolation in ``nd_filtter_calibration.py``. I have kept the same shift in EXCAM pixels, (3,3), by adjusting the FPAM H/V delta values. The previous version had also assumed an identity matrix for Fpam2EXCAM transformations, which is far from the actual value and is not proportional to an identity matrix either.

hsergi, 2025-08-19:
Going to PR it

closed 2025-08-19 by semaphoreP