roman-corgi/corgidrp #327: doesn't match the north position angle

taichiuyama, 2025-03-10:
**Describe the bug**
I got different values between the input north angle value for mocks.create_astrom_data and the extracted north angle from astrom.boresight_calibration.

**To Reproduce**
field_path = os.path.join(“YOUR PATH“, “JWST_CALFIELD2020.csv”)
ang_list = [45,90,135,180]
north_angle=15

for ang in ang_list:
    # make a mock dataset
    mock_dataset = mocks.create_astrom_data(field_path, rotation=north_angle)
    hd=mock_dataset[0].ext_hdr
    CD1_2 = hd[‘CD1_2’]
    CD2_2 = hd[‘CD2_2’]
    roll_angle = np.rad2deg(np.arctan2(CD1_2, CD2_2)) # Compute North Position Angle
    print(roll_angle)

    # run the boresight calibration to get an AstrometricCalibration file
    astrom_cal = astrom.boresight_calibration(mock_dataset, field_path, find_threshold=500)
    print(astrom_cal.northangle)

**Expected behavior**
seems to be the same numbers?

**corgidrp version:**
 - Release or github commit/branch

**Additional context**
I'm making mock files using mocks.create_astrom_data and running the boresight calibration. I got different numbers from each calculation, and wondered if they should be exactly the same value. If different values are fine, how much difference can the mock file input 'north_angle' and 'astrom_cal.northangle' be allowed?

semaphoreP, 2025-03-10:
Can you report in this issue how different the values are?

taichiuyama, 2025-03-10:
it depends on the input parameters (e.g., rotation in mocks.create_astrom_data, and find_threshold in astrom.boresight_calibration).

if I adopt rotation=15 deg and find_threshold=50, I got 14.865 for astrom.northangle, showing 0.135 deg gap

semaphoreP, 2025-06-05:
I don't think so?

semaphoreP, 2026-09-08:
Closing as no longer able to reproduce. I think perhaps some of the new changes have fixed whatever was the underlying error.

closed 2026-09-08 by semaphoreP