roman-corgi/corgidrp #334: Filename Conventions need to be updated to propagate the last filename when files are combined

maxwellmb, 2025-03-17:
When multiple files are combined into a single file, we want to propagate the VisitiID and the FTIMEUTC from the _last_ file of the files that were combined to be the filename of the newly combined file. 

The DRP needs to be scrubbed for cases when this is done. Photon counting, the combine.py functions, and the PSF subtraction step all come to mind, but there might be more. 

There will be a separate issue created specifically about this for calibration files.

maxwellmb, 2025-03-18:
The expected filenames can be found on the following pages: 
[L1](https://collaboration.ipac.caltech.edu/display/romancoronagraph/L1+Current+DRP+Development+Version+-+SSC+Release+3.2)
[L2a/L2b](https://collaboration.ipac.caltech.edu/display/romancoronagraph/L2+DRP+Current+Development+Version) - Note Combining Photon-Counting frames has been identified as one of the places where this Issue is relevant
[L3](https://collaboration.ipac.caltech.edu/display/romancoronagraph/L3+Current+DRP+Development+Version) 
[L4](https://collaboration.ipac.caltech.edu/display/romancoronagraph/L4+Current+DRP+Development+Version) - This is another place where frame combining is relevant (PSF subtraction)

alexisyslau, 2025-03-18:
Before editing few lines, would like to check-in first. Sooooo, under the current defintion: 

https://github.com/roman-corgi/corgidrp/blob/4030f7295173d6f5417beab553d350e9244affd5/tests/test_photon_counting.py#L74-L75

we get `dataset_err` from using function `def create_photon_countable_frames()` from `mocks.py`

https://github.com/roman-corgi/corgidrp/blob/4030f7295173d6f5417beab553d350e9244affd5/corgidrp/mocks.py#L2587-L2600

Question 1: 

Do we just change these line from the photon_counting.py such that 

https://github.com/roman-corgi/corgidrp/blob/4030f7295173d6f5417beab553d350e9244affd5/corgidrp/photon_counting.py#L285-L287

`new_image.filename = CGI_<dataset[-1].pri_hdr['VISITID']>_<rounded number of dataset[-1].ext_hdr['VISITID']>_L2b.fits`? 
While leaving mocks.py and the filenames in mocks.py alone? I guess those filenames are just dummy at the moment?

semaphoreP, 2025-03-19:
I think if the input L2a data already has the same file format, it's just the same as doing `dataset[-1].filename.replace("L2a", "L2b")`

alexisyslau, 2025-03-20:
Cool, I still have couple more questions: 

1. For now, we are assuming that the incoming data already have the **same file format**, meaning that for photon counting, we will change from  ` new_image.filename = dataset[0].filename.split('.')[0]+'_pc'+filename_end+'.fits' ` to `dataset[-1].filename.replace("L2a", "L2b")`.

This can break some tests within test_photoncounting.py but I can change the test to only run when seeing "CGI_" file format. 

2. In combine.py, walker.py is using it. It is not clear to me what file it should be and what file it is going to output, so I am not sure if there is a need to change the filename. Am I missing something somewhere in the drp? 

3. PSF subtraction is in l3tol4.py, is it sufficient to change just add a line in the function `do_psf_subtraction` before returning the dataset, `dataset_out = update_to_l4(dataset_out)`. However, following `ttr5_analogy.py`, the `update_to_l4` is done after the PSF subtraction?

semaphoreP, 2025-03-20:
Yes, some of our tests do not currently follow this filenaming convention. I think this is ok for this release, but we do want to change them all eventually.

semaphoreP, 2025-03-20:
Yes, update_to_l4 and the other (update_to_lx) functions will advance the data level and change the filename. Thus, do_psf_subtraction() does not need to change the filename. combine.py does not either, except that the new filename of the combined file should be the last filename of the files combined. The advancement to L4 will be done by update_to_l4, which is its own step in a recipe.

semaphoreP, 2025-03-21:
Should be accomplished in #341. Re-open if more issues are found

closed 2025-03-21 by semaphoreP