roman-corgi/corgidrp #342: DRP should use ISPC from the extension header instead of PHTCNT

juliamilton, 2025-03-19:
There are two L1 header keywords related to photon counting: "PHTCNT" in the primary header which indicates that photon counting was planned for that observation, and "ISPC" in the extension header which comes from the instrument. 

Functions in the DRP that look for a photon-counting indicator should likely use the actual state of the instrument, rather than the planned state.

nasavbailey, 2025-03-19:
A quick grep shows PHTCNT also shows up in walker.py and some end2end tests. So tagging @semaphoreP for awareness.

More context: The user has the option to select whether to automatically calculate the best EXCAM params (EM gain, exp t, PC flag) based on the actual achieved contrast during HOWFS/GITL. In this case, the original 'planned' values (HDU0 values) are ignored. The telemetered values (HDU1 keywords) always reflect the current state of the instrument, so should take precedence over the 'planned' values (HDU0 keywords) for data processing.  This was a hidden feature, since documentation describing that feature was never scrubbed of export-controlled content and released to CPP.

semaphoreP, 2025-03-19:
This is being fixed in PR #340

nasavbailey, 2025-03-19:
I would also like to request that PHTCNT be updated to communicate when auto-setting is enabled. This is already the case for OPGAIN, which is a string that is either the requested value or AUTO. I suggest we do something similar with PHTCNT, but that would change the datatype from bool to str. Once this issue is closed, we'll have confirmed DRP isn't using PHTCNT, so I don't think a bool->str change would cause any problems for DRP. I've put that request on the L1 change request page and tagged Eric and Cynthia

semaphoreP, 2025-03-24:
Addressed in #340

closed 2025-03-24 by semaphoreP