roman-corgi/corgidrp #344: Update trap pumping header keywords to fit within 8 characters

juliamilton, 2025-03-19:
Current trap pumping-only headers implemented in the DRP, TPSCHEME1-TPSCHEME4, are > 8 characters. Update the code (and the wiki L1 header page) to make them TPSCHEM1-TPSCHEM4. This is already implemented on the SSC side and does not require a change request to them.

Tagging @kjl0025 for awareness.

kjl0025, 2025-04-23:
Hey @juliamilton, I have a PR I'm working on, and I can easily fix this.  Is that cool with you if I lump this in with my PR?  

I'll be adjust header info on Confluence today, and while I'm at it, I can edit these in the L2, L3, and L4 pages (and they are already correct on the trap cal page).  I can let you handle the L1 page since I can't edit that one?

juliamilton, 2025-04-23:
> Hey [@juliamilton](https://github.com/juliamilton), I have a PR I'm working on, and I can easily fix this. Is that cool with you if I lump this in with my PR?
>
> I'll be adjust header info on [...]

That works for me, thank you! I will edit the L1 page.