roman-corgi/corgidrp #345: Hard-coded file paths in e2e tests need to be replaced with something for general usage.

ajeldorado, 2025-03-19:
**Describe the bug**
Several of the e2e test files have the path to the test data hard-coded. Some of the examples I found (in the _develop_ branch) are:

```
tests/e2e_tests/kgain_e2e.py:    tvacdata_dir = '/home/jwang/Desktop/CGI_TVAC_Data/'  
tests/e2e_tests/nonlin_e2e.py:    TVACDATA_DIR = '/home/jwang/Desktop/CGI_TVAC_Data/'
tests/e2e_tests/nonlin_kgain_e2e.py:    TVACDATA_DIR = '/home/jwang/Desktop/CGI_TVAC_Data/'

corgidrp/recipe_templates/l2a_to_l2b_tdd_vap.json:    "outputdir" : "/Users/ygouf/git/cgi-sit/13-tdd/vi_tdd_04_iit_data_processing/output_drp_L2b/",
tests/e2e_tests/astrom_e2e.py:    tvacdata_dir = "/Users/kevinludwick/Library/CloudStorage/Box-Box/CGI_TVAC_Data/Working_Folder/" #"/Users/macuser/Roman/corgi_contributions/Callibration_Notebooks/TVAC"
tests/e2e_tests/bp_map_e2e.py:    # tvacdata_dir = "/Users/jmilton/Documents/CGI/CGI_TVAC_Data"
tests/e2e_tests/bp_map_e2e.py:    tvacdata_dir = "/Users/kevinludwick/Library/CloudStorage/Box-Box/CGI_TVAC_Data/"
tests/e2e_tests/flatfield_e2e.py:    tvacdata_dir = '/Users/kevinludwick/Library/CloudStorage/Box-Box/CGI_TVAC_Data/Working_Folder'
tests/e2e_tests/l1_to_l2a_e2e.py:    tvacdata_dir = "/Users/kevinludwick/Library/CloudStorage/Box-Box/CGI_TVAC_Data/Working_Folder/"#"/Users/jmilton/Documents/CGI/CGI_TVAC_Data"
tests/e2e_tests/l1_to_l2b_e2e.py:    tvacdata_dir =  "/Users/kevinludwick/Library/CloudStorage/Box-Box/CGI_TVAC_Data/Working_Folder/"
tests/e2e_tests/noisemap_cal_e2e.py:    tvacdata_dir = "/Users/kevinludwick/Library/CloudStorage/Box-Box/CGI_TVAC_Data/Working_Folder/"
tests/e2e_tests/photon_count_e2e.py:    tvacdata_dir =  "/Users/kevinludwick/Library/CloudStorage/Box-Box/CGI_TVAC_Data/Working_Folder/"
tests/e2e_tests/trad_dark_e2e.py:    tvacdata_dir = "/Users/kevinludwick/Library/CloudStorage/Box-Box/CGI_TVAC_Data/Working_Folder/"
tests/e2e_tests/trap_pump_cal_e2e.py:    tvacdata_dir = "/Users/kevinludwick/Library/CloudStorage/Box-Box/CGI_TVAC_Data/Working_Folder/"
tests/e2e_tests/trap_pump_cal_e2e.py:        output_dir = r"/Users/kevinludwick/Library/CloudStorage/Box-Box/CGI_TVAC_Data/TV-20_EXCAM_noise_characterization/simulated_e2e_trap_pumped_frames/"
```

**To Reproduce**
Someone other than the user tries to run a *e2e.py script and it fails because of nonexistent paths.

**Expected behavior**
Any user needs to be able to run the e2e test without modifying the file contents.

**corgidrp version:**
 - develop branch

**Additional context**
For GSW R3.0.2 verification testing, we need to be able to call the e2e test files as-is. Maybe there could be two environment variables read in in the `if __name__ == "__main__":` block of the scripts to set values for `tvacdata_dir` and, if it is used, `output_dir`.

semaphoreP, 2025-03-19:
If you look at the README, you'll see that you should be running the e2e tests following the following syntax `pytest --which e2e --tvacdata_path /path/to/CGI_TVAC_Data --e2eoutput_path tests/e2e_tests/ tests/e2e_tests/`. This syntax ignores the `if __name__ == "__main__"` which is used only for debugging.

ajeldorado, 2025-03-19:
Oh, OK. Sounds like a non-issue then.

closed 2025-03-19 by ajeldorado