roman-corgi/corgidrp #356: DRP v2 (CTC R3.0.2) data products need to conform to filename conventions

nasavbailey, 2025-03-21:
All data products delivered in DRP v2 (CTC R3.0.2) need to conform to the filename conventions described in the Data Format pages on their respective pages on the CPP wiki. https://collaboration.ipac.caltech.edu/display/romancoronagraph/Data+Format+and+Volume 

For example L3s: CGI_<VisitID: PPPPPCCAAASSSOOOVVV>_<Last TimeUTC>_L3_.fits

semaphoreP, 2025-03-24:
This issue seems vague. @nasavbailey can you comment on what actions you are looking for?

nasavbailey, 2025-03-24:
Based on discussion in DRP WG telecon today: Update L#, TDA, or CAL FITS file creation so the output filename follows the convention _whenever it's given an input dataset that also obeys the convention._  

For example, for L3->L4: if given a set of L3 filenames that follow CGI_<VisitID: PPPPPCCAAASSSOOOVVV>_<TimeUTC>_L3_.fits, then confirm the L4 filename is CGI_<Last science target VisitID: PPPPPCCAAASSSOOOVVV>_<Last science target TimeUTC>_L4_.fits. There should be a unit test.

Each product owner should do this for their own product. Please check the box when your PR addressing this has been merged. We're collecting all these together in this issue, rather than tracking each in a separate issues.

- [ ] L2a  @semaphoreP  
- [ ] L2b analog  @semaphoreP 
- [ ] L2b Photon Counted @kjl0025 
- [ ] L3 @maxwellmb 
- [ ] L4 PSF-subtracted @maxwellmb 
- [ ] L4 : not PSF-subtracted	same as owner of https://github.com/roman-corgi/corgidrp/issues/362 
- [ ] ABF (absolute flux) @JuergenSchreiber 
- [ ] AST (astrometry) @manduhmia 
- [ ] BPM (bad pixel map) @juliamilton 
- [ ] CTP (core throughput) @hsergi 
- [ ] DNM (detector noise maps) @kjl0025 
- [ ] FLT (flat fields) @semaphoreP 
- [ ] NDF (ND filter sweet spot) @juliamilton 
- [ ] NLN (nonlin calibrations) @hsergi 
- [ ] TPU (chargetrap/trap pumping call) @kjl0025

JuergenSchreiber, 2025-04-01:
I see also other data products like KGAIN which are not in that filename scheme, should they be added?

semaphoreP, 2025-04-01:
Yes, we should add KGAIN

semaphoreP, 2025-04-02:
I believe this is complete. Please re-open if not.

closed 2025-04-02 by semaphoreP