roman-corgi/corgidrp #358: HDU0 keyword ORIGIN is incorrect for DRP-generated products

nasavbailey, 2025-03-22:
HDU0 keyword ORIGIN is not updated by DRP. Instead the L1 value of "SSC" is preserved. This keyword should be updated to reflect who generated the product: us!

nasavbailey, 2025-03-22:
Possible values I can think of: DRP or CTC (because, formally, DRP data products are "delivered" via CTC). I prefer DRP, because CTC may create their own engineering data products, and this would allow us to differentiate who is doing the analysis.

ecady-jpl, 2025-03-22:
I'm OK with using DRP as the default.  There aren't currently any Calibration Reference Data that aren't sourced from the DRP.  CTC has deprecated all of our internal tools to make products when they were moved into DRP, and we wouldn't try to make "shadow products" or anything.

semaphoreP, 2025-03-27:
I am modifying so that when we load any fits file into the DRP, the origin automatically becomes the DRP.

closed 2025-03-28 by semaphoreP