roman-corgi/corgidrp #375: inject PSFs for pyKLIP algorithm throughput which match the FWHM vs radial separation used for core throughput calculation

kjl0025, 2025-03-26:
**Is your feature request related to a problem? Please describe.**
Currently, the PSFs injected into the scene for calculating the pyKLIP algorithm throughput do not necessarily match the FWHM for a given radial separation used for core throughput calculation.

**Describe the solution you'd like**
Make the injected PSFs match.

semaphoreP, 2025-03-27:
Which part are you referring to? I thought we are injecting PSFs straight from the core throughput dataset into the data

kjl0025, 2025-03-27:
Currently the PSFs that are injected have FWHM = 1.22 lam/D, from line 216 of klip_fm.py, in meas_klip_thrupt().

semaphoreP, 2025-03-27:
It's only used in measure noise, to determine annulus sizes, not for injecting planets

kjl0025, 2025-03-27:
Oh, never mind, I agree with you.  We can close this issue.

closed 2025-03-27 by semaphoreP