roman-corgi/corgidrp #401: Suppress all warnings in tests

semaphoreP, 2025-04-09:
**Describe the bug**
Unit tests and e2e tests produce a lot of warnings right now. E.g., the current unit tests produce 626 warnings! We don't want warnings in production.

**To Reproduce**
Run the e2e or unit tests with pytest

**Expected behavior**
We should get 0 warnings.

**corgidrp version:**
 - v2.0

**Additional context**
Break up this issue into smaller issues in the future.

semaphoreP, 2025-06-05:
I would say, unless it's obvious why the behavior produces the warning and how to fix it, we can just suppress the warning. However we should suppress warnings on a line-by-line basis, and not across an entire file. That way, we only suppress the warnings we are expecting.

maxwellmb, 2025-06-06:
Modules

wbalmer, 2025-07-08:
Wrapped up the work on the unit tests, moving to e2e warnings now.

everlastingEric, 2025-08-27:
@semaphoreP Yes I have noticed that too. I believe those are all deprecation warnings caused by os.fork(), as I couldn’t find any other warnings when I run the unit tests. I could try to find where it is raised and suppress them for now but since those are deprecation warnings that probably something that needs to eventually be fixed.

semaphoreP, 2025-08-27:
Thanks. Would be nice for someone to track what is calling os.folk() and whether this is something in our control. You are free to work on it or we can cycle other people on it, if you are focusing on pol mode.

everlastingEric, 2025-08-29:
I took a look at the deprecation warnings again, I'm pretty sure they are coming from somewhere inside the pyklip package and that's where os.fork() gets called. Not sure if there's anything to do to fix that if it's not coming from a corgidrp module so for now I can just try to locate the exact lines of code and supress it.

everlastingEric, 2025-08-29:
For your reference I think this is one of the places where it pops up: 
```
File "/home/ericshen/corgi/corgidrp/corgidrp/klip_fm.py", line 311, in meas_klip_thrupt klip_dataset(pyklip_dataset, outputdir=klip_params['outdir'], File "/home/ericshen/miniconda3/envs/corgi/lib/python3.12/site-packages/pyklip/parallelized.py", line 1744, in klip_dataset
```
I'm not too familiar with pyklip but I think it comes up when it gets called with multiple threads

everlastingEric, 2025-08-29:
ok so all of the remaining deprecation warnings coming from `test_flux_ratio_noise_curve.py`, `test_klipthrupt.py`, `test_measure_companions.py`, and `test_psfsub.py` all stem from this one place in the stack trace shown in the comment above 
```
  File "/home/ericshen/miniconda3/envs/corgi/lib/python3.12/site-packages/pyklip/parallelized.py", line 766, in rotate_imgs
    tpool = mp.Pool(processes=numthreads)
```
I don't know if there's a good way to suppress those within corgidrp with just `warnings.catch_warnings()` since they are coming from different threads, but if you can suppress/fix it from within pyklip that might be ideal. Other than that, I do see one more warning from the new spec unit test that I can suppress real quick and I'll make sure nothing else comes up in the e2e tests.

semaphoreP, 2025-09-02:
Are you saying you're trying to catch the warning in the corgidrp step functions (e.g., psf_subtraction()) and that the child processes don't see that? The other option is to do `mp.set_start_method("spawn")` at the beginning of each test function.

semaphoreP, 2025-09-02:
I don't know about you, but I don't actually get the warnings in my console (e.g., if I can the test function directly). Only when I run it with pytest does pytest find that the warning exists.

everlastingEric, 2025-09-02:
Yeah same here, it only seems to show up in pytest

everlastingEric, 2025-09-03:
Update: all of the existing unit test warnings should be gone now, including the deprecation warnings. I also caught and suppressed a few more warnings in the e2e tests, but as of right now, I can't guarantee all of the e2e test warnings are caught. I think we should probably go through the e2e tests once more using pytest once all of the e2e tests are fixed up and working. For now, I'll make a PR with the changes I've made so far.