roman-corgi/corgidrp #413: East angle isn't currently computed for star fields

ajeldorado, 2025-04-25:
**Is your feature request related to a problem? Please describe.**
The TDD FDD says that the L2b->L3 pipeline will do/compute these: "construct WCS:
distortion correction, convert from DN to electrons, plate scale, North and
East angles". `astrom.py` computes the plate scale and north angle but not the east angle.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
`astrom.py` needs to compute the east angle right? Or will all images always already be oriented with the cardinal directions as one expects to see them (i.e., without a mirror flip)?

**Additional context**
- Needed for R.3.0.2

semaphoreP, 2025-04-25:
I believe that we should know from optical modeling whether east should be 90 degrees CW or CCW of North. I believe we determined east should always be 90 degrees CCW of North. Perhaps @nasavbailey remembers when this was determined? 

I'll note that `create_wcs` in l2_to_l3.py is the function that actually constructs the WCS. And the CD matrix it creates uniquely specifies the North and East angles.

semaphoreP, 2025-05-06:
Do we need to do anything for this, or can we close this issue?

nasavbailey, 2025-05-06:
https://collaboration.ipac.caltech.edu/spaces/romancoronagraph/pages/136120763/Operational+Activities+Design+Document+-+OADD I don't know if the issue can be closed, but I did add the x/y vs N/E diagram from the OADD to the CPP wiki

nasavbailey, 2025-06-05:
I still see it in the latest release of the TDD FDD (Rev G), but someone (@mygouf ?) must have determined that the current implementation is sufficient. So I would say that unless @mygouf objects, we close this one.

Paper trail ... the test definition document (JPL internal reference: "VI: L5 TDD 6 - GSW") does include a check for reported E angle.  The VAP should have been written to implement that test definition. And the VAP test report does not say there was a failure (JPL internal reference: CGIAR-1085 ). I haven't gone back to read the VAP code itself to see what specifically was tested, but I don't think it's worth digging further.

maxwellmb, 2025-07-24:
@nasavbailey said we can close this.