roman-corgi/corgidrp #414: Revert plate scale to be a scalar instead of being different in x and y

ajeldorado, 2025-04-25:
**Is your feature request related to a problem? Please describe.**
A very recent ECR just changed the wording of REQ 1159971 to specify that only a single value of plate scale in mas/pixel will be computed over the whole image. However, at the same time, `astrom.py` was changed from doing that in v1 to computing separate values along x and y in v2 to meet the deprecated wording of the requirement. After talking to Max, it sounds like everyone prefers the single value for plate scale.

**Describe the solution you'd like**
Revert the computation in astrom.py to compute just a single value of plate scale, with any discrepancy along the axes showing up in the distortion fit instead.

**Additional context**
Needed for R.3.0.2

semaphoreP, 2025-04-25:
Assigning to @manduhmia, since she implemented this change.

What's the timeline for getting this in?

ajeldorado, 2025-04-25:
The current schedule gives us about 2 weeks to have everything packaged and tested and working, so maybe a week at most for all fixes?