roman-corgi/corgidrp #418: Hardcoded calspec file list does not match list of absflux calibrators and always requires external download

ecady-jpl, 2025-04-30:
**Describe the bug**
Two concerns:

1. [Hardcoding the list of absolute flux calibrators](https://github.com/roman-corgi/corgidrp/blob/v2.2/corgidrp/fluxcal.py#L18-L36) means that they cannot be updated in ground software outside of a software release cycle.  Already the list differs from [the list being maintained by the Observation Planning group in the targets wiki](https://cpptargs.sioslab.com/index.php/Calibration_Targets#Absolute_Flux).
2. The calibrators are downloaded from a [hardcoded STScI URL](https://github.com/roman-corgi/corgidrp/blob/v2.2/corgidrp/fluxcal.py#L38) every call to `fluxcal.get_calspec_file()`.  This is vulnerable to both link rot and to lack of connectivity when using corgidrp.  Might be less of a big deal if `get_calspec_file()` wasn't called in other places (e.g. [every time `determine_app_mag()` in `l4_to_tda.py` is called](https://github.com/roman-corgi/corgidrp/blob/v2.2/corgidrp/l4_to_tda.py#L67).)

**Expected behavior**
It would be preferable if the list of calibrators was able to be updated outside of the software release cycle, and if no corgidrp commands required an external web connection to function.  If there is a particular finite set of data files that are getting repeatedly downloaded (e.g. Vega), it might be better to Git-LFS them with the repo rather than keep pulling them.

**corgidrp version:**
 - v2.2

JuergenSchreiber, 2025-05-02:
In l4_to_tda.determine_app_mag() and l4_to_tda.determine_color_cor() both ways are possible: you either input the name of the calspec star and retrieve the corresponding fits file from the url or you already have a calspec fits file and directly input this. 
In case of the flux and also the nd filter calibration it is indeed relying on the stars name (target header) and retrieve the calspec fits file from the internet. 
I think also here, we can easily adjust this and add an additional parameter, so that the user also can input a spectral model fits file that he wants.

ecady-jpl, 2025-05-02:
My concern with the external internet connection is not it getting overloaded, but it existing at all.  corgidrp isn't standalone if it requires a specific URL at STScI to continue to exist.

Call this a lesson learned: I have personally been on an observing run and had a piece of software we were about to use fail to start in the middle of the night because it couldn't connect to a license server.  Requiring unnecessary external connections can make the software implementation fragile.

juliamilton, 2025-05-06:
> Sounds like a good thing to work on for v3, but I don't see it as urgent. It would be good to discuss with the calibration team on revisiting that list of calibration targets for abs flux calibration and how large that list will end up being.
> 
> I'm not really worried about L4 to TDA, because these steps aren't intended to be run by CTC. TDA analysis is expected to be done by the CPP, given the open-ended nature of TDA analysis. The bigger worry would be that these calls are used in the production of the absolute flux and ND filter calibrations. We should really make sure we have a clear strategy of how those calibrations will run if we observe a star that is not on the list of calspec stars. Pinging [@JuergenSchreiber](https://github.com/JuergenSchreiber) and [@juliamilton](https://github.com/juliamilton) for comment (question for you two: can the recipes for these cal files by edited to handle stars not in the current list)?

Wanted to clarify that the recipes for the ND filter cal product (and flux cal as Juergen said) can handle any calibrator with a CALSPEC file that exists either locally or online, but are not set up to handle stars that aren't in calspec. At the minimum, we need an expected flux for a star in a specific bandpass to do these calculations.

maxwellmb, 2025-06-11:
this solves part of the problem: instead of hardcoding the calspec list in `fluxcal.py`, can we store it in a .csv file or something. That way we can consider it a "config file" that we can update without being tied to a software release.

JuergenSchreiber, 2025-06-12:
What is the conclusion now?
I like the .csv file idea of Max. But I do not think that we should store all corresponding CALSPEC data in DRP. If this is clarified , I would volunteer to assign me on this.

semaphoreP, 2025-06-13:
@JuergenSchreiber can you explain what is the purpose of having the list of all possible target stars? Can't we just check if a star's CALSPEC file exists in .corgidrp/calspec_data or could be queried online on the fly? 

For calspec filepath, is it possible for both `calibrate_fluxcal_aper()` and `create_nd_filter_cal()`? Those are the step functions exposed to us in recipes.

JuergenSchreiber, 2025-06-16:
Ok, I  see, I have to refactor this. I have the filepath input interface for l4_to_tda.determine_app_mag and l4_to_tda.determine_color_cor, but not for the flux cal determination. It seems I was to lazy and only looked there for the header target name.
I think, we should add this filepath interface there, and second check if the calspec file is in .corgidrp/calspec_data and only download from internet if none of both works. 
The list of all possible target stars is there to have the association to the calspec filename. Should we generate this list as a text file to .corgidrp/calspec_data and read it there?  So one could add a "new" target star there, without having to generate a new corgidrp version? On the other hand, if  people can input the calspec fits file directly, they can look up and download the calspec file by themselves and we do not need this association in this case.

semaphoreP, 2025-06-27:
I believe this has been implemented now in PR #471. Please re-open if something is missing.

closed 2025-06-27 by semaphoreP