roman-corgi/corgidrp #422: Disconnect between corgidrp/__init__.py and the dorba interface

ecady-jpl, 2025-05-02:
**Describe the bug**
There are a couple problems with how the corgidrp initialization in `__init__.py` interacts with dorba.

- The config folder is hardcoded to always [place](https://github.com/roman-corgi/corgidrp/blob/v2.2/corgidrp/__init__.py#L15-L16) and [look for](https://github.com/roman-corgi/corgidrp/blob/v2.2/corgidrp/__init__.py#L64) the config file in the user's home directory (`pathlib.Path.home()`).  Due to the way dorba is set up, that directory will generally not be accessible to it.  There is no path to changing it via dorba, and when this is deployed in the DAE, we likely want to be able to point it at an approved common config rather than in someone's home directory (don't want pipeline behavior for final products to depend on the calling user).
  - As a corollary, functionality that relies on that config folder path (e.g. [here](https://github.com/roman-corgi/corgidrp/blob/v2.2/corgidrp/fluxcal.py#L56) or [here](https://github.com/roman-corgi/corgidrp/blob/v2.2/tests/test_fluxcal.py#L111)) cannot succeed when run with dorba.
- There are several global variables ([here](https://github.com/roman-corgi/corgidrp/blob/v2.2/corgidrp/__init__.py#L60-L61), [here](https://github.com/roman-corgi/corgidrp/blob/v2.2/corgidrp/__init__.py#L14)) which are assumed to be persistent.  However, these are not accessible via dorba to change manually (would require a setter/getter as dorba can't access class attributes that aren't methods) or by config file (due to the issues described above).   They also do not persist between calls to functions in the docker image unless run in batch mode.
  - Upshot is that some of the variables in that file cannot be changed from their defaults, which means some saving functionality will not work (e.g. [here](https://github.com/roman-corgi/corgidrp/blob/v2.2/corgidrp/caldb.py#L375-L384)), some tests cannot succeed (e.g. tests using `track_individual_errors` as [here](https://github.com/roman-corgi/corgidrp/blob/v2.2/tests/test_fluxcal.py#L247-L248)), and we will not have access to the [`skip_missing_cal_steps`](https://github.com/roman-corgi/corgidrp/blob/v2.2/corgidrp/__init__.py#L74) flag when running the DRP on the DAE as it defaults to False.

**Expected behavior**
It may be necessary to modify how the global configuration parameters are stored and set in corgidrp.  I think if at minimum there were a keyword argument to [`update_pipeline_settings()`](https://github.com/roman-corgi/corgidrp/blob/v2.2/corgidrp/__init__.py#L56) that let a user supply the path to the config file via function argument, and temporarily load the globals from there, we may be able to get it work as expected.

**corgidrp version:**
 - v2.2

ecady-jpl, 2025-05-02:
No, I didn't know about it.  That would be great if it worked.  Let me try...

semaphoreP, 2025-05-02:
Regarding global variables, you won't be able to change them in DOBRA, running line-by-line some of the unit tests that modify them won't work. But, is that something you want to do? Those pipeline settings are generally advanced things that we don't expect to normally use in production. 

You can change global variables by passing in json recipes, where you specify how you want to set certain variables. Not sure if that's useful to you.

ecady-jpl, 2025-05-02:
Hmm.  Looking at them now: is the expectation that you call `step_2_load_cal(this_caldb=step_1_initialize(), main_cal_dir='/path/to/dorba/accessible/cfg')` to end up with a caldb with all the local settings you want?

And to your question: we're trying to use the corgidrp test codes as a basis for V&V activities.  It's possible some of the steps being checked are low-enough level that they don't fold into requirement verification, we'll evaluate.  Maybe we can get around it.  But I would at some point (not in this thread) want to chat with you + Max + probably Marie and/or Vanessa to make sure I understand your expected "in production" use, especially for the calibration products.

semaphoreP, 2025-05-05:
Hi Eric, yes, I think that's the intention. For unit testing, you probably can just point main_cal_dir to an empty directory. 

I'll note that you probably only need a subset of corgidrp tests for V&V. We have written a lot of tests for optional features/flexbility in the DRP that suits the interests of the community team. For example, you probably don't ever need to touch `skip_missing_cal_steps`.

maxwellmb, 2025-06-05:
@ecady-jpl just checking in on this. Is this something we still need to resolve or do those ops functions help?

ecady-jpl, 2025-06-05:
Let's close this up for now.  We managed to get what we needed to work during V&V activities.  If there's a specific change in the future, I'll open a new issue for it.

closed 2025-06-05 by ecady-jpl