roman-corgi/corgidrp #424: Write hlc instead of hcl in l2b_to_l4_e2e tests

hsergi, 2025-05-05:
The following tests have `hcl` hardcoded instead of the usual `hlc`. This may induce some confusion when another user downloads the OS11 data and does not misspel `hlc`: https://roman.ipac.caltech.edu/data/sims/coronagraph/os11/hlc_os11_v3.zip 

```
l2b_to_l4_e2e.py:    input_hdul = fits.open(os.path.join(e2edata_path,"hcl_os11_v3",input_file))
l2b_to_l4_e2e.py:    data = np.loadtxt(os.path.join(e2edata_path,"hcl_os11_v3",'hlc_os11_batch_info.txt'), skiprows=2)
l2b_to_l4_e2e.py:    #This folder should contain an OS11 folder: ""hcl_os11_v3" with the OS11 data in it.
l2b_to_l4_noncoron_e2e.py:    input_hdul = fits.open(os.path.join(e2edata_path, "hcl_os11_v3", input_file))
l2b_to_l4_noncoron_e2e.py:    #This folder should contain an OS11 folder: ""hcl_os11_v3" with the OS11 data in it.
```

bensutlieff, 2025-05-22:
I've fixed this in the above pull request (#447).

semaphoreP, 2025-08-29:
Closed in #447