roman-corgi/corgidrp #425: Report SED/ spectral type of detected source

juliamilton, 2025-05-06:
**Is your feature request related to a problem? Please describe.**
L4 -> TDA processing specifies that the assumed spectral type of the target star and each identified companion should be reported. Currently this is not implemented. 

**Describe the solution you'd like**
Report assumed spectral type of target star and companions in final TDA output from CALSPEC or other means.

maxwellmb, 2025-05-22:
Discussed briefly in DR&S telecon May 22: we should evaluate if we want to actually do this or accept it as a permanent ECR change. That question is still open, but we should consider it.

maxwellmb, 2025-05-22:
After discussions with Me/Jason/Marie, we decided that outputting the SED was fine and that's easy because we already need that, we just need to output it.