roman-corgi/corgidrp #427: Spectroscopy: Estimate dispersion angle and fit dispersion profile

semaphoreP, 2025-05-06:
**Is your feature request related to a problem? Please describe.**
Estimate dispersion angle and fit dispersion profile. Use as input the centroids computed by `spec.compute_psf_centroid()` (see PR #399). 

**Describe the solution you'd like**
Fit the dispersion parameters to input centroids using estimate_dispersion_clocking_angle() and fit_dispersion_polynomials(). Output a “Dispersion Calibration” class (does not necessarily need to be saved)

JuergenSchreiber, 2025-05-14:
To better understand: Who else is also working on the spectroscopy branch? The stuff in spectroscopy.py is from Neil and should be ported to spec.py in corgi format? I  see some functions like Gaussian1/2d , centroiding, which I think we should import from astropy and/or photutils?

semaphoreP, 2025-05-14:
@Ezarsh02 worked on the first step (measuring centroids in prep for measuring dispersion clocking angle and dispersion profile), but is finishing her semester-long internship, so we have no one else currently.

JuergenSchreiber, 2025-05-15:
@semaphoreP thanks Jason for the info, should I then work on on the spectroscopy-test branch?

semaphoreP, 2025-05-15:
That was just used for the dev work that has already been PR'd back into spectroscopy. So, you should feel free to branch off of spectroscopy however you want. Thanks! Can I assign you to this issue?

JuergenSchreiber, 2025-05-15:
OK, I see, I was too fast, I am already assigned to it.

semaphoreP, 2025-05-15:
Great! Implementation note: I think you can take as input a `SpectroscopyCentroidPSF` instance, instead of the dictionaries in the original spectroscopy.py code.

JuergenSchreiber, 2025-05-21:
@semaphoreP Hi Jason, I saw that the spectroscopy branch is more than 1000 commits behind main, which might cause some problems. Should I pull main in or (develop?) first before going on?

semaphoreP, 2025-05-21:
Pull in develop. Thanks!

JuergenSchreiber, 2025-06-02:
@semaphoreP Hi Jason, it seems that in spec.compute_psf_centroid() the psf template is not considered. Should I stay with this or change it? Well, to change it we would have to consider an additional template dataset as input. I am not really clear whether I understand it, but I think the template input would be some model PSF to be a fit model beside a 2D Gaussian in addition.

hsergi, 2025-07-21:
NOTE: The CFAM filter image offsets will be measured during commissioning. There should be an option to load them If you need to consider the CFAM offsets in this task. The future requirement in the spectroscopy functional design documentation (FDD) will be something like "CFAM Filter Offsets. ID-TBD - The CTC GSW should provide a table of relative image shifts (x,y) on EXCAM between filters due to the filter wedge angles."

semaphoreP, 2025-08-01:
Implemented in #472

closed 2025-08-01 by semaphoreP