roman-corgi/corgidrp #428: Spectroscopy: Create Wavelength Solution

semaphoreP, 2025-05-06:
**Is your feature request related to a problem? Please describe.**
Create Wavelength Solution (wavelength-to-pixel map) and save as calibration file

**Describe the solution you'd like**
Using Dispersion Calibration and an input zero-point coordinate (manual for now potentially(?) -- look into automating it), make a Wavelength Solution with create_wave_cal_map().

**Additional context**
Needs issue #427 to produce dispersion calibration
See test_spectroscopy.py `test_wave_cal` function for guidance.

hsergi, 2025-05-19:
Notice that the source position needs to be estimated by a combination of target knowledge, star location on EXCAM and EXCAM orientation and pixel scale (see also Neil's comments on Spectroscopy FDD on confluence: [link](https://collaboration.ipac.caltech.edu/spaces/romancoronagraph/pages/154410149/Spectroscopy+FDD+development+area)).

If this information is not captured yet (as a Header keyword, for instance), use a place holder. For instance, input keyword parameters in the step function. Consult!

semaphoreP, 2025-05-21:
> Notice that the source position needs to be estimated by a combination of target knowledge, star location on EXCAM and EXCAM orientation and pixel scale (see also Neil's comments on Spectroscopy FDD [...]

Note that this issue just creates the wavelength calibration file, not solve for the wavelength of any arbitrary science data.

hsergi, 2025-06-06:
This is part of the FDD AAC requirements, which are reviewed and accepted. [AAC](https://collaboration.ipac.caltech.edu/spaces/romancoronagraph/pages/106725541/JPL+Functional+Design+Descriptions#JPLFunctionalDesignDescriptions-SpectroscopicCalibrations). Work can start w/o any further delay

hsergi, 2025-07-15:
NOTE: this is an L3->L4 step in the Data Processing pipeline