roman-corgi/corgidrp #429: Spectroscopy: Measure LSF

semaphoreP, 2025-05-06:
**Is your feature request related to a problem? Please describe.**
Measure LSF and save as calibration file

**Describe the solution you'd like**
Port over fit_line_spread_function() from `spectroscopy.py` into a step function in `spec.py`, save LSF as a new calibration data class. 

**Additional context**
Needs output of #428 
See test_spectroscopy.py `test_line_spread_function` for guidance (in spectroscopy branch).

JuergenSchreiber, 2025-05-08:
Where can I find spectroscopy.py etc.?

hsergi, 2025-06-05:
This is part of the FDD AAC requirements, which are reviewed and accepted: https://collaboration.ipac.caltech.edu/spaces/romancoronagraph/pages/106725541/JPL+Functional+Design+Descriptions#JPLFunctionalDesignDescriptions-SpectroscopicCalibrations. Work can start w/o any further delay