roman-corgi/corgidrp #436: test_nonlin_cal.py defines teardown_module() twice

ajeldorado, 2025-05-08:
**Describe the bug**
test_nonlin_cal.py defines teardown_module() twice, so it is not clear which is being used. The test runs, but this should be fixed to avoid any potential problems in the future.

```
git grep  'teardown_module' ./ 
test_calibrate_darks_lsq.py:def teardown_module():
test_calibrate_darks_lsq.py:    teardown_module()
test_corethroughput.py:def teardown_module():
test_err_dq.py:def teardown_module():
test_kgain_cal.py:def teardown_module():
test_nonlin_cal.py:def teardown_module():
test_nonlin_cal.py:def teardown_module():
```

**To Reproduce**
Just open tests/test_nonlin_cal.py to see the double definition.

**Expected behavior**
Just one definition of teardown_module()

**corgidrp version:**
 - v2.2

**Additional context**
N/A

semaphoreP, 2025-05-08:
It's the second one that get used (the first definition is overridden), but we should fix this.

bensutlieff, 2025-05-22:
I think I've fixed this - see the pull request above (#446). Also, the first definition was indeed overridden by the second definition, but the second definition wasn't deleting all of the variables defined in the setup_module. I've kept the one that correctly deletes all of the variables.

semaphoreP, 2025-06-05:
Fixed in PR #446

closed 2025-06-05 by semaphoreP