roman-corgi/corgidrp #444: Re-evaluate data-quality / bad pixel logic for stacked/combined frames

juliamilton, 2025-05-21:
In several pipelines (e.g. photon_counting.py, combine.py) the DQ extension for a stacked product is generated by bitwise-OR-ing the DQ masks of the input frames. This guarantees that any pixel marked bad in any contributing frame will be flagged in the output, but it may be overly conservative when:

- A transient defect affects only a subset of the frames and is averaged out in the stack.

- Down-stream users would benefit from knowing why a pixel is flagged (e.g. saturation vs. cosmic-ray hit vs. no-photon detection), which is lost when all bad reasons collapse to a single 1.

Request:
- Review existing code that builds DQ for combined frames (start with photon_counting.py, combine.py); list their current logic.

- Define and implement new logic as needed

maxwellmb, 2025-05-21:
I'm not sure about photon_counting.py, but I think `combine_images` in combine.py already does what you want.

semaphoreP, 2025-05-21:
Yes, I think this might be a problem for photon_counting but I don't think either issue is a problem for combine.py (it only keeps the DQ flag when ALL values going into the combine of a pixel have a DQ flag raised; the final DQ flag for those with non-zero DQ flag is sticky, so all DQ reasons are still preserved in the final combined frame).

closed 2025-07-18 by semaphoreP