roman-corgi/corgidrp #445: Spectroscopy: create SPC CT correction

hsergi, 2025-05-21:
**Is your feature request related to a problem? Please describe.**
The L3 --> L4 step for spectroscopy [(link)](https://collaboration.ipac.caltech.edu/spaces/romancoronagraph/pages/154410149/Spectroscopy+FDD+development+area) requires to provide the CT correction when the SPC mask has been used to correct the attenuation of the planet PSF

**Describe the solution you'd like**

A step function that estimates the SPC CT correction. Technical details need to be clarified in the spectroscopy TDD FDD or in a section of Confluence devoted to the implementation of spectroscopy

**Additional context**

hsergi, 2025-05-23:
Updated Context above

hsergi, 2025-06-05:
This is part of the FDD AAC requirements, which are reviewed and accepted: https://collaboration.ipac.caltech.edu/spaces/romancoronagraph/pages/106725541/JPL+Functional+Design+Descriptions#JPLFunctionalDesignDescriptions-SpectroscopicCalibrations. Work can start w/o any further delay

hsergi, 2025-06-05:
The core throughput calibration is without prism (see PAM configuration on [Confluence](https://collaboration.ipac.caltech.edu/spaces/romancoronagraph/pages/106725541/JPL+Functional+Design+Descriptions#JPLFunctionalDesignDescriptions-SpectroscopicCalibrations)).

Hence, the existing code for CT calibration in the Imaging mode should be resusable. One has to check that there's nothing specific a PAM selection on it (like CFAM) that would rpevent the code from working with Band 3 (or Band 2). 

Notice that for e2e tests we may want to consider a similar pattern as it will be used with the SPC (TBD). Contact @neilzim for more info.

semaphoreP, 2025-08-14:
Note that @MalachiNoel3 is working on this, but I can't add him to the assignee.