roman-corgi/corgidrp #448: Spectroscopy: ND Filter calibration

semaphoreP, 2025-05-22:
**Is your feature request related to a problem? Please describe.**
We need to check and adapt the ND filter calibration code to be able to do ND filter calibration for spec mode observations. The ND filter transmission needs to be measured as a function of wavelength. 

The baseline configuration is: the FPAM ND225 filter, calibrated in Band 3 with the prism in.

**Describe the solution you'd like**
Adapt the existing ND filter calibration code, but we will need to measure the OD vs wavelength. This will require changing the ND filter calibration class, or making a new one to handle this different data format.

hsergi, 2025-07-22:
Confirmed: The fulfillment of the Absolute Flux and ND filter calibration requirements as a function of wavelength may be accomplished after release R3.1. Tim Koch, CTC's lead is fine with having these requirements created now, and open for R3.1, to be closed later on.

hsergi, 2025-07-30:
Once this task is addressed, one should include the need to update the Absolute FLux calibration as a function of wavelength too. The Absolute Flux and ND filter calibration are requirements CGI-REQT-5476, CGI-REQT-5477 and CGI-REQT-5478 in the Spectroscopy Observations FDD

hsergi, 2025-08-14:
For R3.1, we are going to use the ND filter and Absolute calibration tools used for broadband imaging, see https://github.com/roman-corgi/corgidrp/issues/492

JuergenSchreiber, 2025-12-05:
In the Spectroscopy Observations Functional Design Description it is stated that the ND filter is not necessary for non-coronagraphic spectrophotometric standard star observations, is this correct? Do we need then the spectral ND filter calibration at all?

semaphoreP, 2026-02-09:
@agreenbaum This is the new calibration data type we haven't done yet. It will have a `DATATYPE = "NDSpectroscopy"` but we have not implemented it yet. It should be very similar to the `DATATYPE = "NDFilterSweetSpotDataset"` data format, but we won't have it done for the great header release.