roman-corgi/corgidrp #450: Update L1 file name and header & propagate changes

agreenbaum, 2025-05-23:
**Is your feature request related to a problem? Please describe.**
File names need to be changed to lowercase, by request of SOC/MAST and the L1 header keywords are receiving a small update (additional values)

**Describe the solution you'd like**
The new file name format needs to be updated to lowercase letters throughout all data products. New header keywords should be added to L1 model. It probably only makes sense to propagate "DETECTOR" if any of the new keywords.

**Additional context**
Three new header keywords: 
DETECTOR (always set to EXCAM)
FILENAME (name of the file)
FSWDVERS (version of CGI FSW dictionary used to process the telemetry)
See: https://collaboration.ipac.caltech.edu/spaces/romancoronagraph/pages/161617086/L1+Current+DRP+Development+Version+-+SSC+Release+3.3.0

ecady-jpl, 2025-06-24:
Does this include the capital T in the middle of the datestamp?

agreenbaum, 2025-06-26:
Not for SSC current release, no. I do need to check with SOC if that is ok or if we'll have to make another change.

maxwellmb, 2025-07-30:
@agreenbaum, any updates on the T?

agreenbaum, 2025-07-30:
Thanks for the reminder. We'll need to make the "T" lowercase to conform with SOC.