roman-corgi/corgidrp #465: Spectroscopy L2b->L3: Image wavelength calibration map for the estimated source position

hsergi, 2025-06-02:
**Is your feature request related to a problem? Please describe.**
Construct image wavelength calibration map for the estimated source position

**Describe the solution you'd like**

The calibration step (#428) produces a wave cal map. It does not apply the last step of shifting the map to the source coordinates. That is, the calibration step function will not use any information about the star position or source offset, just the unocculted star + prism + filter calibration data.

The shift of the map to the source coordinates depends on the science data. The calibration map should be usable by multiple science datasets.

**Additional context**
The calibration map is dealt within #428

hsergi, 2025-06-02:
To define a spectroscopy campaign, CPGS will somehow need to set the astrometric offset of the target, in order to program the FSAM position. Check if the SSC/CTC team have defined that interface. Depending on how that information is propagated, that would determine sensible default position (after converting to L3 EXCAM pixel coordinates). We would also want the user to have the option to specify an arbitrary angular separation and position angle.

hsergi, 2025-07-30:
https://github.com/roman-corgi/fsampos does the translation of delta X/Y EXCAM pixels into delta FSAM H/V values in microns

JuergenSchreiber, 2025-08-15:
I think the title should be changed to L3_to_L4, Essentially this issue will be solved with #476 together with #428