roman-corgi/corgidrp #467: Document RAM footprint of PSF subtraction for a realistic data size

semaphoreP, 2025-06-05:
**Is your feature request related to a problem? Please describe.**
We want to understand RAM/CPU time use for a realistic PSF subtraction dataset (let's say 1000 science target data, all cropped, and 500 reference star images).

**Describe the solution you'd like**
Adapt the L3_to_L4 test code to do this. We can duplicate frames if needed to get to 1000 frames (The accuracy of the result doesn't matter, we just want to understand resource footprint).

**Additional context**
Putting it in a R3.1 milestone, but TBD if this is the right milestone

everlastingEric, 2025-08-09:
@maxwellmb No I have not with a bigger dataset, I will do that as well.

everlastingEric, 2025-08-12:
PSF subtraction RAM data can be found on the same page as the photon counting RAM data: https://collaboration.ipac.caltech.edu/spaces/romancoronagraph/pages/173181245/Corgi-DRP+RAM+Usage+Report

semaphoreP, 2025-08-14:
Closing this issue. Please see #491 for discussions on actions to reduce data size.

closed 2025-08-14 by semaphoreP