roman-corgi/corgidrp #476: Spectroscopy Cal: Wavelength zero-point solution

hsergi, 2025-07-14:
**Is your feature request related to a problem? Please describe.**

Measure the centroid of the DM satellite spot.

**Describe the solution you'd like**
The wavelength zero-point solution is a per-observation quantity. Add the x,y coordinates in EXCAM pixel units to some new keywords in the processed images.

**Additional context**
This is the first part of the "wavelength solution". The requirement for the wavelength zero-point is: " Given (1) a cleaned image of a prism-dispersed DM satellite spot observed through the FSAM slit mask and a narrowband calibration filter, (2) the relative image shifts between bands due to filter wedge angles, CTC GSW should compute the wavelength zero-point."

The "wvelength solution" is dealt with in task https://github.com/roman-corgi/corgidrp/issues/428

hsergi, 2025-07-15:
NOTE: this is an L3->L4 step in the Data Processing pipeline

JuergenSchreiber, 2025-07-29:
In course of development of the wavelength calibration in branch spec_wavecal I have added a WavelengthZeropoint calibration class to data.

semaphoreP, 2025-07-30:
What is the difference between this task and #465? Are they the same thing (with #465 being old enough that it wasn't moved to L3 -> L4)?

semaphoreP, 2025-08-04:
Hmm... good question. I think saving them into the observations should be ok, unless @hsergi knows we HAVE to save the wavelength solution as a separate calibration file for SSC. 

One thing to note: only a small subsets of an observation with DM spots turned on are used for zeropoint. And that wavelength solution derived from those few frames needs to apply to the rest of the data.

JuergenSchreiber, 2025-08-05:
Do I understand this right that we produce a wavelength calibration file with a "reference" observation (described in #465) and apply and shift this wavelength map to the actual observed stars zero point as described in #428?

JuergenSchreiber, 2025-08-06:
Then it would make sense to add the wavelength + error map as extensions to the Frames, although this would increase the memory used?  Maybe better to just carry this product on (without saving a fits file) to the spectral extraction, where we then save the wavelength scale anyhow in the final spectrum product? Anyway, the wavelength map would not be a calibration file.
I guess we have to select the SATSPOTS = 1 frames first and then fit the zero point?

semaphoreP, 2025-08-06:
> Then it would make sense to add the wavelength + error map as extensions to the Frames, although this would increase the memory used? Maybe better to just carry this product on (without saving a [...]

I think the memory addition is not huge. this is not a worry. You are correct that we need SATSPOTS=1 frames to find zero-point. This will likely require new simulated data, so let's maybe (for the first PR) just have the function to create the wavelength solution assuming a measured zero-point, and then we work on the function to measure the zeropoint. 

It does look like there is a requirement to produce a wavelength solution, but to me, it seems like saving the wavelength solution as a FITS extension of the science data is one way to satisfy this requirement (instead of saving it as a separate calibration file). Let me know what you think Sergi and Neil.

semaphoreP, 2025-08-06:
The wavelength map will not have a dedicated class if we store it as a FITS extension.

mygouf, 2025-08-07:
As long as it doesn't introduce complications on Julia's side (VAP testing), which I don't think it's case based on what I heard from Julia, I'm fine with storing the wavelength map as a FITS extension to the processed prism image.

semaphoreP, 2025-08-28:
Implemented in #505

closed 2025-08-28 by semaphoreP