roman-corgi/corgidrp #481: Add safeguard for frames whose pixel values exceed detector saturation but lack the OVEREXP flag

juliamilton, 2025-08-05:
In L1-> L2a processing, the `detect_cosmic_rays` step identifies pixels as being saturated if they are above 70% of the detector's FWC (FWC from the detector-params file = approx 11,000 ADC counts, with a kgain of 8.7). Frames where a significant number of pixels are above this value but below the EXCAM's over-illumination threshold (16,000 ADC counts) do not trigger the OVEREXP header flag. Because the flag is absent, the DRP accepts the data, yet the cosmic-ray module still classifies nearly every pixel as saturated. Processing those frames through this step takes significant time and likely would produce unusable outputs.

**Issue:**

The DRP has no mechanism to abort processing, or switch to an alternate path, when a large fraction of pixels are above the detector’s saturation threshold while OVEREXP == 0. This allows saturated frames to propagate through the pipeline.

**Suggested fix:**

Do a pre-processing check that computes the fraction of pixels whose values exceed some “saturation threshold” (default = 70 % of FWC from the detector-params file).

If that fraction exceeds a configurable limit, either:

(1) Abort the recipe and raise an exception that lists the file with the issue, or

(2) Flag saturated pixels as bad, skip cosmic-ray detection, and continue.

semaphoreP, 2025-08-05:
our `frame_selection` step in l2a_to_l2b does indeed have an optional flag to filter on the fraction of bad pixels in an image, and flag as bad when there is too many: 

[image].

Is this sufficient?

juliamilton, 2025-08-05:
> I agree that flagging those pixels with DQ values would be good.
> 
> There's a step where we filter out bad files, we could add a flag for X number of bad pixels.
> 
> Why do we need to skip the cosmic ray step?

I think in general it would be useful to just not process a frame if a high fraction of pixels are saturated, since the algorithm spends a lot of time iterating on data that will all be flagged anyway. Perhaps a reason to skip the cosmic ray step and keep processing would be for flat field images, or other situations where the field is intentionally bright.

juliamilton, 2025-08-08:
> Do we really need automatic safeguards for this, or if it's just for calibration data, can we just adjust parameters in our recipes as it comes up?

I'm going to defer to @nasavbailey and @ecady-jpl on that. This has come up as an issue in testing but not sure how likely we'd be to encounter saturated frames in actual usage.

mygouf, 2025-08-13:
> Do we really need automatic safeguards for this, or if it's just for calibration data, can we just adjust parameters in our recipes as it comes up?

I'm not sure to understand your question Jason, can you elaborate on this?

semaphoreP, 2025-08-19:
I think we can certainly add this to the list of things we'd like to do, but maybe for a v4 release then. I'd like to make sure we stay focused on our v3 goals in the next 2 months. 

I'll note that for commissioning, there's a good chance we'll be manually processing data in addition to the automated processing (this happened quite often at the start of JWST, same with GPI comissioning). So, I don't think we should think of the automated processing as the only thing we have to depend on (but certainly would be good to do as much of it automatically as possible!). Also, the L1->L2a processing is so fast right now, that even a 100x slowdown doesn't really impact processing time.

ecady-jpl, 2025-08-19:
Well, it's technically possible to parallelize the lower-level pipelines, yes.  But that would have to be a very deliberate choice in the non-DRP outer wrapper tools to make that happen, and we've been hitting resource limitations as it is.  This does seem like it's out of scope of this ticket.

semaphoreP, 2025-08-19:
I don't understand why it'd be so hard to parallelize calls to a python script. I would naively think this would be straightforward to do this (e.g., gnu parallel). But agreed, out of scope of this issue.

mygouf, 2025-08-20:
> I think we can certainly add this to the list of things we'd like to do, but maybe for a v4 release then. I'd like to make sure we stay focused on our v3 goals in the next 2 months.

Agreed on this. And it seems Eric is on the same page as well on that point as he doesn't anticipate this to be an issue over the next ~2 months. I suggest we agree on a plan to fix this so that when it's time to tackle it we have a path forward. What about flagging saturated frames at the beginning of L1 processing and skipping the rest of the processing for those frames and proceed as usual for non-saturated frames?

semaphoreP, 2025-08-20:
Yes, I think we want another step function similar to `frame_select()` in l2a_to_l2b but for l1_to_l2a, and only focused on throwing away frames too saturated.

ecady-jpl, 2025-08-20:
What happens to frames that get "thrown away"?  They never get processed to L2a, or they just end up with all pixels marked as bad?

semaphoreP, 2025-08-20:
We can do either. (frame_select defaults to discarding them, so they don't make it to the next data level, but can be toggled to keep them with a IS_BAD = True flag). It sounds like in this case, we do want to discard them, since the main concern is that the next processing steps in L1->L2a are too slow when there are too many saturated pixels.

silverSapphire, 2025-09-19:
I have a draft solution for this issue, but I need clarification on what it would mean to "discard frames" in this context. Can @semaphoreP (or other) review my commit and indicate the desired behavior?

semaphoreP, 2025-09-19:
Basically remove them from the dataset for future processing. I'm looking for the same implementation as what's done for discarding frames in `frame_selection()` in l2a_to_l2b.py (https://github.com/roman-corgi/corgidrp/blob/main/corgidrp/l2a_to_l2b.py#L245)