roman-corgi/corgidrp #492: Spectroscopy Cal: Test current ND filer and Absolute calibration with Bands 3F and 2F

hsergi, 2025-08-14:
**Is your feature request related to a problem? Please describe.**
Spectroscopy requires to have a wavelength dependent ND and Absolute flux calibration. This capability will be implemented after Milestone R3.1 (https://github.com/roman-corgi/corgidrp/issues/448). For R3.1, we will use the current functionality developed for broadband imaging.

**Describe the solution you'd like**
Test whether the current functionality developed for broadband imaging also works if Band 1F is replaced by 3F or 2F and if the current broadband functions have anything setup to only work with HLC Band 1 imaging only that could prevent them to work with spectroscopic data.

hsergi, 2025-08-19:
@semaphoreP Here's an analysis of what current functions do. It does not seem that there's anything that could conflict with spectroscopy frames. I wonder which kind of additional test would be worth adding here. 

``fluxcal.py``: HLC, IMAG or Imaging do not show at all in any place. FPAM,/fpam FSAM/fsam, LSAM /lsam and SPAM/spam are not used in fluxcal.py. cfam is not used. CFAM is used in fluxcal.get_filter_name(). CFAM=3F or 2F must load the corresponding filters because they are present in corgidrp/data/filter_curves/. Minor bit to fix: The variable ``filter_name``, which reads the value of CFAM, in l.503, right before the call to get_fillter_name() in the function calibrate_fluxcal_aper() is not used. I would delete that line.

``test_fluxcal.py``: HLC, IMAG or Imaging do not show at all anywhere in the function. FPAM is set once to ``ND475`` for the mock data in the test. CFAM is set by default to ``3C`` (already a different value than band 1). CFAM is also set to ``5C`` to create a test that will fail.

Can you share your thought about which test could be suitable to address this issue? One option is to create some mock data as in the tests of the functions below, but with the headers corresponding to a spectroscopy configuration. Then, run the same tests as in the main branch.

I also reached out to @juliamilton to get any additional feedback from her experience when she desalt with these functions in the previous release.

semaphoreP, 2025-08-19:
It would be good to copy the existing e2e tests for nd and abs flux (they're both pretty small and easy to duplicate), change the simulated data so the headers use the band 3 headers (like CFAMNAME), and make sure the test passes. I just want to convince ourselves we have nothing to worry about when running our existing code on another filter. As you mentioned, we only need to process the ND filter and abs flux calibrations in regular imaging mode (But in Band 3), so we don't need to worry the code not handling spectral data -- the ND and abs flux cal frames we feed in will be imaging, just in a different filter.

hsergi, 2025-08-19:
I think there is an inconsistency in `develop` on the way the filename for `FluxcalFactor' is handled (lower/upper case). `Main` is fine -since it passed all the e2e tests. 

[fluxcal_e2e.py](https://github.com/roman-corgi/corgidrp/blob/d0c52ae3e145ea945d6dc260a897212b643bd78d/tests/e2e_tests/fluxcal_e2e.py#L50)

[data.py](https://github.com/roman-corgi/corgidrp/blob/d0c52ae3e145ea945d6dc260a897212b643bd78d/corgidrp/data.py#L1611)

Is there a preference for this issue? "abf_cal" or "ABF_CAL".  I based my current branch on develop so far.

semaphoreP, 2025-08-19:
This is because PR #480 moved all filenames to lowercase (new SSC convention), but did not do it for the e2e tests. please fix the e2e tests as you encounter them for your development. I think @kjl0025 or @juliamilton will make a big push to fix all the e2e tests soon.