roman-corgi/corgidrp #501: Fix numpy deprecation warnings

semaphoreP, 2025-08-20:
**Describe the bug**
Two deprecations (both in two locations)

```
"l3_to_l4.py", line 530, in do_psf_subtraction
    sci_dataset = split_datasets[int(np.nonzero(np.array(unique_vals) == 0)[0])]
                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
DeprecationWarning: Conversion of an array with ndim > 0 to a scalar is deprecated, and will error in future. Ensure you extract a single element from your array before performing this operation. (Deprecated NumPy 1.25.)
```

```
"l3_to_l4.py", line 537, in do_psf_subtraction
    ref_dataset = split_datasets[int(np.nonzero(np.array(unique_vals) == 1)[0])]
                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
DeprecationWarning: Conversion of an array with ndim > 0 to a scalar is deprecated, and will error in future. Ensure you extract a single element from your array before performing this operation. (Deprecated NumPy 1.25.)
```

```
"calibrate_kgain.py", line 562, in calibrate_kgain
    np.sqrt(np.square(np.reshape(std_diffs[x], 
                      ^^^^^^^^^^^^^^^^^^^^^^^^
  File "site-packages/numpy/_core/fromnumeric.py", line 314, in reshape
    warnings.warn(
DeprecationWarning: `newshape` keyword argument is deprecated, use `shape=...` or pass shape positionally instead. (deprecated in NumPy 2.1)
```

```
"calibrate_kgain.py", line 510, in calibrate_kgain
    np.sqrt(np.square(np.reshape(std_diffs[x], 
                      ^^^^^^^^^^^^^^^^^^^^^^^^
  File "site-packages/numpy/_core/fromnumeric.py", line 314, in reshape
    warnings.warn(
DeprecationWarning: `newshape` keyword argument is deprecated, use `shape=...` or pass shape positionally instead. (deprecated in NumPy 2.1)
```

**To Reproduce**
Steps to reproduce the behavior:

 * l2b_to_l4_e2e.py for the first two
 * kgain_e2e.py for the last two

**Expected behavior**
A clear and concise description of what you expected to happen.

Run with no warnings

**corgidrp version:**
 - develop

SaiashishChauhan-Infinium, 2025-08-20:
I looked into the deprecation warnings. They are due to recent NumPy updates (1.25+ and 2.1).

Fixes applied:

In l3_to_l4.py (lines 530, 537), explicitly extract the scalar index:
**`sci_dataset = split_datasets[int(np.nonzero(np.array(unique_vals) == 0)[0][0])]`
`ref_dataset = split_datasets[int(np.nonzero(np.array(unique_vals) == 1)[0][0])]`**

In calibrate_kgain.py (lines 510, 562), update reshape call to remove deprecated newshape keyword:
**`np.sqrt(np.square(np.reshape(std_diffs[x], (-1,))))`**

After these changes, the warnings may resolve when running l2b_to_l4_e2e.py and kgain_e2e.py.

_Hope it solves your problem :)_

wbalmer, 2025-08-20:
I remember these, I think I corrected them in some other places in #473 . Once the e2e error fixes are merged I could quickly fix these.

wbalmer, 2025-08-20:
okay! I can PR from [small_deps_fix](https://github.com/roman-corgi/corgidrp/tree/small_deps_fix)

semaphoreP, 2025-08-22:
@wbalmer , just noticed, is the deprecation on line 510 of kgain not addressed? There were two deprecation warnings in kgain

semaphoreP, 2025-08-25:
Ok, closing for now then.