roman-corgi/corgidrp #543: Spectroscopy VAP: L4->TDA Spectroscopy slit transmission map

juliamilton, 2025-09-15:
**REQT:** CGI-REQT-5475
**Purpose:** Compute 2-D slit transmission from matched L4 grids with slit-in vs slit-out.

**Inputs:**

1. L4 (slit-in) dataset spanning a 3x3 FSM grid:

- CFAMNAME identical across images
- FSAMNAME = SLIT
- Headers include unique (FSMX, FSMY)

2. L4 (slit-out) dataset spanning a 3×3 FSM grid:

- CFAMNAME identical & matching slit-in set
- FSAMNAME = OPEN
- Headers include unique (FSMX, FSMY) matched to slit-in dataset

**Procedure (high-level):**
See issues for details: https://collaboration.ipac.caltech.edu/spaces/romancoronagraph/pages/173180475/R3.1+VAP+Planning

1. Validate inputs (levels, data type, headers, filenames).
2. Validate both grids, register/ interpolate if needed.
3. Compute transmission.
4. Write output TDA file to disk.
5. Log checks and validate outputs (levels, data type, headers, filename) using functions in check.py.

**Output:**
Slit transmission map (TDA product)

hsergi, 2025-10-07:
The current set of tests are documented in [VAP testing R3.1](https://collaboration.ipac.caltech.edu/spaces/romancoronagraph/pages/173180475/R3.1+VAP+Planning)

hsergi, 2025-10-07:
A summary:

```

- Input slit-in images (L4 DRP dataset):

Check dimensions.
Check DATALVL = L4.
Verify all images have identical CFAMNAME values.
Verify all images have FSAMNAME = SLIT.
Verify all images have(FSMX, FSMY) values.
Filename convention: cgi_<visitid>_<yyyymmddthhmmsss>_l4_.fits

- Input slit-out images (L4 DRP dataset):
Check dimensions.
Check DATALVL = L4.
Verify all images have identical CFAMNAME values (same as slit-in set).
Verify all images have FSAMNAME = OPEN.
Verify all images have(FSMX, FSMY) values.
Filename convention: cgi_<visitid>_<yyyymmddthhmmsss>_l4_.fits

- Output slit transmission map (TDA product):
Check dimensions.
Confirm slit transmission contains: 
Slit transmission map values with as many wavelength steps as input data.
Locations associated to each slit transmission value with as many values as slit transmission values per wavelength step

- Baseline performance checks:
Print peak and mean transmission values.
```