roman-corgi/corgidrp #545: Spectroscopy VAP: L4->TDA Registered prism image of star

juliamilton, 2025-09-15:
**REQT:** CGI-REQT-5465
**Purpose:** Identify the dispersed unocculted-star image whose PSF-to-FSAM slit alignment best matches the target source; write an L4 table with the best FSM offsets & filename.

**Inputs:**

1. L4 (slit-in) dataset spanning a 3x3 FSM grid:

- CFAMNAME identical across images
- FSAMNAME = SLIT
- Headers include unique (FSMX, FSMY)

2. Processing function keywords:

- xcent_template, ycent_template (template source estimate)
- slit_align_err (alignment error from FSAM slit)

**Procedure (high-level):**
See issues for details: https://collaboration.ipac.caltech.edu/spaces/romancoronagraph/pages/173180475/R3.1+VAP+Planning

1. Validate inputs (levels, data type, headers, filenames).
2. Compute alignment error vs template; select best-matching image.
3. Write output TDA file to disk.
4. Log checks and validate outputs (levels, data type, headers, filename) using functions in check.py.

**Output:**
Registered prism image table (TDA product)

hsergi, 2025-09-22:
This is the actual, updated battery of VAP tests: CGI-REQT-5465 [link](https://collaboration.ipac.caltech.edu/spaces/romancoronagraph/pages/173180475/R3.1+VAP+Planning)

**Input images (L2b files)**:

- Template data:
  - Confirm presence of alignment offsets relative to FSAM slit in all file headers
  - Confirm presence of WV0_X and WV0_Y in all file headers
- FSM data:
  - Check DATALVL = L2b.
  - Filename convention: cgi_<visitid>_<yyyymmddthhmmsss>_l2b.fits
  - Verify all images have identical CFAMNAME values.

**Output L2b filenames:**

- Check data type:

DATALVL = L2b

- Filename convention: cgi_<visitid>_<yyyymmddthhmmsss>_l2b.fits
- Confirm filenames are part of the input L2b FSM files

**Baseline performance checks:**

- Print best-matching filenames

hsergi, 2025-09-26:
Implemented in #564 . Can this be closed? Or should we wait until Julia runs the interim VAP testing?