roman-corgi/corgidrp #552: Ensure old e2e tests generate mock calibrations with valid CFAMNAME, DPAMNAME, and FPAMNAME values

semaphoreP, 2025-09-16:
**Is your feature request related to a problem? Please describe.**
Our e2e tests (in particular, l2b_to_l4_e2e.py, there may not be others, but we should check) that require corethroughput, abs flux, or ND filter calibration need to generate mock calibrations with valid CFAMNAME, DPAMNAME, and FPAMNAME for the caldb. 

**Describe the solution you'd like**
Update e2e tests to generate mock calibrations for corethroughput (and if applicable, abs flux and ND filter) to initialize these 3 keywords to reasonable values. 

In caldb, set `filter_calib(..., err_if_none=False)` for all calls and make sure e2e tests still pass.

semaphoreP, 2025-09-16:
See discussion in PR #536 if more context is needed. This doesn't break the pipeline in any way, but we should be more specific in picking calibrations automatically.

juliamilton, 2025-09-18:
> I believe the changes I'm making in [#488](https://github.com/roman-corgi/corgidrp/pull/488) will address this

Update: #557 partially addresses this (makes sure that all calibrations have valid values for CFAMNAME, DPAMNAME, and FPAMNAME), but needs further work to make sure the values are logically consistent with the PAM values that should be set during the VISTYPE associated with that e2e test.

semaphoreP, 2025-12-18:
This seems too general now that we've mostly fixed it, and we should make specific issues as we identify in the header cleanup.